wil3/gymfc #1: Windows installation

zbenic, 2018-05-24:
Hello.

If I manage to install [Gazebo on Windows](http://gazebosim.org/tutorials?tut=install_on_windows), would Gym FC work on Windows 10?

wil3, 2018-05-24:
It should. The front end uses Python3 and the backend is using Gazebo 8 so it should work on Windows. If you give it a shot please keep me posted here on your progress or if you need further help, I don't have Windows to test it on. I also have some baseline scripts I can add to the repo so you know if its working properly.

zbenic, 2018-06-02:
Hello again,

So, after a few painful days. I have managed, I guess, to install Gazebo 8 on Win10.
After that, I have installed gymfc.

I would very much appreciate to have some scripts to check out the validity of my installations.

wil3, 2018-06-04:
Hey,
Was installing Gazebo on Windows not straight forward? In the future maybe it would be easier to use an Ubuntu virtual environment. If there was anything that may help others install it on windows could you mention them here or make a PR for the README.md?

I pushed the scripts I mentioned and updated the README with how to run it. If everything is good you should get a graph similar to the one I uploaded. I still need to clean up the scripts I uploaded but I just tested them and they work, check the imports for required libs you need, Ill push a requirements.txt when I have time.

I'm going to close this issue, if anything comes up as an issue with GymFC please open a new one.

wil3, 2018-07-04:
Hey @zbenic I forgot gymfc isnt supporting other OSs other than Ubuntu due to how variables are loaded (see #4). I don't have Windows but if you could help me identify how they are set in Windows we could probably get this working.

wil3, 2018-07-04:
Inspecting this further, looking at the Gazebo source the Windows equivalent of `setup.sh` seems to be `win_addpath.bat`. This must be executed before `gzserver` is started

zbenic, 2018-07-04:
Hello @wil3.

I will try again this weekend with the Windows installation. I have not been running that .bat before starting gzserver. I've just try to start gzserver, and it would not (not sure anymore what was the specific problem).

In the meantime, I have setup Ubuntu environment and everything is working just fine.

Just for an information, I've created new world where quadcopters translations are not constrained. 
I'm writing PID controller that will later be used as reference.
Also, I've added contact sensor for the purposes of ending RL episodes (when quadcopter crashes).

But expect more questions coming your way. :)

wil3, 2018-07-04:
Yea normally you dont need to run the setup.sh script (or bat equivalent) before running gzserver. This is only needed if you are changing the environment paths for loading worlds and models from different locations (which GymFC does).

OK sounds good, well if you ever want to try getting it running on Windows, comment back here and well try and get it running. 

Sounds very cool, are you going to teach it to hover and maintain stability? Look forward to hearing more!

wil3, 2018-10-30:
Closing because windows install apparently is still under development.

closed 2018-10-30 by wil3