wil3/gymfc #2: Could not find a version that satisfies the requirement pygazebo

zhaoxinyizhf, 2018-06-10:
Hi, thank you for your inspiring work.
I try to install gymfc with "sudo pip3 install -e".
But I came across some errors:

" Could not find a version that satisfies the requirement pygazebo (from gymfc==0.0.1) (from versions: 2.2.1-2014.1, 2.2.1-2014.2, 3.0.0-2014.1)
No matching distribution found for pygazebo (from gymfc==0.0.1)"

I checked the gazebo version with "dpkg -l | grep gazebo":

"ii  gazebo7-common                                              7.8.1-1~xenial                                        all          Open Source Robotics Simulator - Shared files
ii  gazebo8                                                     8.1.1-1~xenial                                        amd64        Open Source Robotics Simulator
ii  gazebo8-common                                              8.1.1-1~xenial                                        all          Open Source Robotics Simulator - Shared files
ii  gazebo8-plugin-base                                         8.1.1-1~xenial                                        amd64        Open Source Robotics Simulator - base plug-ins
ii  libgazebo8:amd64                                            8.1.1-1~xenial                                        amd64        Open Source Robotics Simulator - shared library
ii  libgazebo8-dev:amd64                                        8.1.1-1~xenial                                        amd64        Open Source Robotics Simulator - Development Files
ii  ros-kinetic-gazebo8-msgs                                    2.5.10-1xenial                                        amd64        Message and service data structures for interacting with Gazebo from ROS.
ii  ros-kinetic-gazebo8-ros                                     2.5.10-1xenial                                        amd64        Provides ROS plugins that offer message and service publishers for interfacing with Gazebo through ROS."

Any clue as to what to I am doing wrong? Thank you again.

wil3, 2018-06-11:
Hi @zhaoxinyizhf thank you for your interest in this project.

The pygazebo project appears to be abandoned. I guess I was using a modified local copy. I went ahead and included this in the repo. Can you please do a pull and try again?
 
Also its easiest to check gazebo version with,
```
gazebo --version
```

wil3, 2018-06-20:
Since there hasn't been any follow up I'm assuming everything is cool so I'm going to close the issue, please reopen if needed.

closed 2018-06-20 by wil3