wil3/gymfc #4: OpenGL Ogre Error

nrocketmann, 2018-07-01:
I am using a mac: OSX version 10.13.2 and I'm trying to install gymfc. I followed the install instructions, and brew installed Gazebo and Ogre. I am doing this in an anaconda3 environment with python3.5.

When I entered the command: `sudo python3 -m gymfc.controllers.iris_pid_eval --env-id=AttFC_GyroErr-MotorVel_M4_Ep-v0`

I got this error:

```
Exception occured connecting to Gazebo retrying .... [Errno 61] Connect call failed ('127.0.0.1', 11345)
Gazebo multi-robot simulator, version 8.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.113.38.81
[Err] [RenderEngine.cc:663] EXCEPTION: unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [RenderingIface.cc:38] Failed to load the Rendering engine subsystem
unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.
[Err] [gazebo.cc:87] Unable to load sensors
[Err] [Server.cc:298] Unable to load gazebo
libc++abi.dylib: Pure virtual function called!
Exception occured connecting to Gazebo retrying .... [Errno 61] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 61] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 61] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 61] Connect call failed ('127.0.0.1', 11345)
Traceback (most recent call last):
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/gazebo_env.py", line 287, in _connect
    self.manager = await pygazebo.connect((self.host, self.gz_port)) 
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 381, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/tasks.py", line 310, in _wakeup
    future.result()
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 294, in result
    raise self._exception
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/pygazebo/pygazebo.py", line 562, in handle_connect
    future.result()
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 294, in result
    raise self._exception
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/pygazebo/pygazebo.py", line 252, in callback_impl
    future.result()  # check for exception
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 294, in result
    raise self._exception
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/tasks.py", line 242, in _step
    result = coro.throw(exc)
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/selector_events.py", line 451, in sock_connect
    return (yield from fut)
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 381, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/tasks.py", line 310, in _wakeup
    future.result()
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 294, in result
    raise self._exception
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/selector_events.py", line 481, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 61] Connect call failed ('127.0.0.1', 11345)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/controllers/iris_pid_eval.py", line 180, in <module>
    main(args.env_id, args.seed)
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/controllers/iris_pid_eval.py", line 163, in main
    env = gym.make(env_id)
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/site-packages/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/attitude_control_env.py", line 67, in __init__
    super(GyroErrorESCVelocityFeedbackEnv, self).__init__(motor_count = motor_count, world=world)
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/gazebo_env.py", line 145, in __init__
    self._start_sim()
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/gazebo_env.py", line 240, in _start_sim
    self.loop.run_until_complete(self._connect())
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/base_events.py", line 467, in run_until_complete
    return future.result()
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/futures.py", line 294, in result
    raise self._exception
  File "/Users/nameerhirschkind/anaconda3/envs/machine-learning/lib/python3.5/asyncio/tasks.py", line 240, in _step
    result = coro.send(None)
  File "/Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/gazebo_env.py", line 293, in _connect
    if not self.manager:
AttributeError: 'GyroErrorESCVelocityFeedbackEnv' object has no attribute 'manager'
```

Any idea on how to fix this?
Thanks in advance!

wil3, 2018-07-01:
Hi @nrocketmann the error you are seeing is from Gazebo, it appears you do not have the proper dependencies. You can confirm this independent of GymFC  from the command line just start Gazebo by typing,
`gazebo`. The only OS I have tested it on is Ubuntu so I dont know how to solve your specific OS problem. My advice would be to try and see how to get Gazebo running on OSX, possibly report an issue to that project is you believe it is a bug. If this is to much of a hassle install Ubuntu in a VM like VirtualBox and run it from in there. I have also heard of success with using Gazebo in Docker. If you can get Gazebo running and you come across a bug in GymFC please let me know. Thanks!

nrocketmann, 2018-07-01:
Thanks wil3. The strange thing is that gazebo ran fine when I ran it from terminal. It may be that I have multiple versions of it or something. I'll run it in virtualbox then with an ubuntu instance.
Thanks!

nrocketmann, 2018-07-03:
Hi @wil3,
Thanks for the tips and advice. I used gazebo8 the first time, which I thought might have been the problem. I tried what you said, and it does indeed look really simple, but I get the same error now, and when I tried `gazebo --version` It says I'm running 8.6. Also, when I type gazebo, it works fine and launches the editor.
Thanks in advance!
Nameer

wil3, 2018-07-04:
Hi @nrocketmann,
You know what, this is my mistake I do think this is an issue with GymFC sorry about that. I forgot it is setting global variables to load Gazebo here, https://github.com/wil3/gymfc/blob/master/gymfc/envs/gazebo_env.py#L218-L223 so its really only compatible with Ubuntu at the moment. GymFC must source the setup file to set global variables, Its currently setting the variables in this file  so if your paths don't match it wont work. This is also related to issue #1. 

Looking closer at Gazebo, it seem to generate setup.sh at install and I don't even know what its doing in Windows (maybe an equivalent bat file). Can you locate your equivalent setup.sh file and let me know where it is located? On Ubuntu it is at `/usr/share/gazebo-8/setup.sh`. I need to work on the best fix for this to be cross-platform. In the meantime to get it working you can use the file paths in your setup.sh for the paths in gazebo_env.py to see if it will work.

nrocketmann, 2018-07-04:
Thanks for the advice! I changed the paths and that does seem to have fixed the previous error, however, I'm now encountering another one:

`[Err] [Plugin.hh:180] Failed to load plugin libQuadcopterWorldPlugin.dylib: dlopen(libQuadcopterWorldPlugin.dylib, 9): image not found`

I tried copying the gymfc assets and plugins into the appropriate gazebo directories, which helped fix an ealier `FileNotFoundError` but it still seems unable to find the `libQuadcopterWorldPlugin.dylib` file. I saw in the `build` directory in gymfc plugins a file with the same name with a `.so` extension instead of `.dylib`. Are these related? If so, how could I fix the issue?

Thanks!
Nameer

wil3, 2018-07-04:
Great to hear that error went away. What startup script did you get the correct paths on OSX from? I want to get this fixed.

However don't move anything out of GymFC. The export variables are to tell Gazebo to look for the assets in GymFC before starting. If you getting a filenotfounderror there's something wrong, post the full error you are getting so we can get that fixed. 

Yep that error is because the plugin has been compiled for Ubuntu. You'll need to compile it for Mac. Gazebo uses cmake to build the plugins, the instructions for Ubuntu are here, https://github.com/wil3/gymfc/tree/master/gymfc/envs/assets/gazebo/plugins just delete the build directory first. I'm using cmake 3.5.2. 

Hope that helps. Let me know how it works out.

nrocketmann, 2018-07-05:
I got the latest cmake (3.12) and tried to compile with the given instructions. I first had to get zmq using these: https://github.com/JustinTulloss/zeromq.node/issues/283 instructions and install uuid, which was easy. Now, I'm getting these other errors that I don't really understand:

````
[ 50%] Building CXX object CMakeFiles/QuadcopterWorldPlugin.dir/QuadcopterWorldPlugin.cpp.o
In file included from /users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.cpp:49:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/sensors/sensors.hh:27:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/sensors/WirelessTransmitter.hh:22:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/physics/Actor.hh:26:
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:159:40: error: 
      reference to 'Spline' is ambiguous
      private: mutable ignition::math::Spline *positionSpline;
                                       ^
/usr/local/include/ignition/math4/ignition/math/Spline.hh:38:33: note: 
      candidate found by name lookup is 'ignition::math::v4::Spline'
    class IGNITION_MATH_VISIBLE Spline
                                ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:28:11: note: 
      candidate found by name lookup is 'ignition::math::Spline'
    class Spline;
          ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:162:40: error: 
      reference to 'RotationSpline' is ambiguous
      private: mutable ignition::math::RotationSpline *rotationSpline;
                                       ^
/usr/local/include/ignition/math4/ignition/math/RotationSpline.hh:34:34: note: 
      candidate found by name lookup is 'ignition::math::v4::RotationSpline'
    class IGNITION_MATH_VISIBLE  RotationSpline
                                 ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:29:11: note: 
      candidate found by name lookup is 'ignition::math::RotationSpline'
    class RotationSpline;
          ^
/users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.cpp:332:29: error: 
      no matching conversion for functional-style cast from 'double' to
      'boost::posix_time::milliseconds' (aka
      'subsecond_duration<boost::posix_time::time_duration, 1000>')
  ...boost::this_thread::sleep(boost::posix_time::milliseconds(msPeriod));
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate
      constructor (the implicit copy constructor) not viable: no known
      conversion from 'double' to 'const
      boost::date_time::subsecond_duration<boost::posix_time::time_duration,
      1000>' for 1st argument
  class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                             ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate
      constructor (the implicit move constructor) not viable: no known
      conversion from 'double' to
      'boost::date_time::subsecond_duration<boost::posix_time::time_duration,
      1000>' for 1st argument
/usr/local/include/boost/date_time/time_duration.hpp:286:59: note: candidate
      template ignored: disabled by 'enable_if' [with T = double]
                                typename boost::enable_if<boost::is_inte...
                                                          ^
3 errors generated.
make[2]: *** [CMakeFiles/QuadcopterWorldPlugin.dir/QuadcopterWorldPlugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/QuadcopterWorldPlugin.dir/all] Error 2
make: *** [all] Error 2
````

wil3, 2018-07-05:
Looks like its an issue with Boost, what version are you using?

nrocketmann, 2018-07-06:
Hi, 
sorry it took so long to get back. I was getting the `zeromq` error again and this time fixed it this way: https://github.com/caskformula/homebrew-caskformula/issues/10  . Now I can reproduce the error I mentioned above.
It looks like I have boost version 1.67.0. I installed it both via brew and macports. Brew also says boost is linked. Should I install a different version of it?

wil3, 2018-07-06:
1.58, not sure why your install is using a newer version. The sleep API changed in newer versions, see if rolling back to 1.58 fixes it.

nrocketmann, 2018-07-06:
I tried installing boost 1.58 with macports, and 1.57 with brew (brew doesn't have 1.58), and still got the same error. Maybe cmake can't find boost? I'm not really sure. I also don't know what paths cmake uses, so I can't change those.

nrocketmann, 2018-07-10:
Yeah, I have been clearing the build directory each time.
Alright, now the boost part of the error is gone. I think brew installing boost did work, I just didn't notice that the error had changed slightly:

```
Scanning dependencies of target QuadcopterWorldPlugin
[ 50%] Building CXX object CMakeFiles/QuadcopterWorldPlugin.dir/QuadcopterWorldPlugin.cpp.o
In file included from /Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.cpp:49:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/sensors/sensors.hh:27:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/sensors/WirelessTransmitter.hh:22:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/physics/Actor.hh:26:
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:159:40: error: 
      reference to 'Spline' is ambiguous
      private: mutable ignition::math::Spline *positionSpline;
                                       ^
/usr/local/include/ignition/math4/ignition/math/Spline.hh:38:33: note: 
      candidate found by name lookup is 'ignition::math::v4::Spline'
    class IGNITION_MATH_VISIBLE Spline
                                ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:28:11: note: 
      candidate found by name lookup is 'ignition::math::Spline'
    class Spline;
          ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:162:40: error: 
      reference to 'RotationSpline' is ambiguous
      private: mutable ignition::math::RotationSpline *rotationSpline;
                                       ^
/usr/local/include/ignition/math4/ignition/math/RotationSpline.hh:34:34: note: 
      candidate found by name lookup is 'ignition::math::v4::RotationSpline'
    class IGNITION_MATH_VISIBLE  RotationSpline
                                 ^
/usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/Animation.hh:29:11: note: 
      candidate found by name lookup is 'ignition::math::RotationSpline'
    class RotationSpline;
          ^
In file included from /Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.cpp:53:
In file included from /Users/nameerhirschkind/PycharmProjects/drone/gymfc/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.hh:21:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/common.hh:12:
In file included from /usr/local/Cellar/gazebo8/8.6.0/include/gazebo-8/gazebo/common/CommonIface.hh:22:
/opt/local/include/boost/uuid/sha1.hpp:13:9: warning: This header is
      implementation detail and provided for backwards compatibility.
      [-W#pragma-messages]
#pragma message("This header is implementation detail and provided for b...
        ^
1 warning and 2 errors generated.
make[2]: *** [CMakeFiles/QuadcopterWorldPlugin.dir/QuadcopterWorldPlugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/QuadcopterWorldPlugin.dir/all] Error 2
make: *** [all] Error 2
```
It looks to me like a problem with the ignition math library: https://ignitionrobotics.org/libs/math
Any ideas?
Thanks!

wil3, 2018-07-15:
Could be another version conflict, compare the library versions to those installed on on Gazebo 8 on Ubuntu to what you have installed on your system.  What have you tried so far?

weixingsun, 2018-11-15:
sun$ python3 -m gymfc.controllers.iris_pid_eval --env-id=AttFC_GyroErr-MotorVel_M4_Ep-v0
Traceback (most recent call last):
  File "/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/sun/Desktop/OpenAI/gymfc/gymfc/controllers/iris_pid_eval.py", line 7, in <module>
    from mpi4py import MPI
**ModuleNotFoundError: No module named 'mpi4py'**

wil3, 2018-11-15:
Hi @weixingsun when you do `pip3 freeze | grep mpi4py` what is the output?

wil3, 2018-11-15:
Sorry this issue should have been closed, I don't have any plans to support OSX. Unless your Gazebo paths match the startup in https://github.com/wil3/gymfc/blob/master/gymfc/envs/gazebo_env.py#L242 it isn't going to work. Version 2 will tie into Gazebo without hardcoded paths making it more flexible but I don't know if its enough. If you do get it working submitting a pull request for any documentation and changes would be great.

closed 2018-11-15 by wil3

ssdasgupta, 2019-05-30:
@nrocketmann , did you solve the problem finally?

nrocketmann, 2019-05-30:
@ssdasgupta, I'm afraid I didn't. I haven't looked at this in about a year now. Best of luck if you're going down the same path.