wil3/gymfc #6: Other RL Controllers

leonardoaraujosantos, 2018-07-06:
Hi @wil3 you will include other types of RL controllers like PPO/TRPO?

Thanks

lengjia, 2018-07-09:
I have the same problems in using baselines algorithms of PPO/TRPO, can you give the source code of RL controllers?

wil3, 2018-07-09:
Hi @leonardoaraujosantos and @lengjia , please read our paper and the README file, this is a gym environment, not a repo for controllers. Our work used the OpenAI baselines repo to construct RL controllers. The PID controller is only given as a means to ensure the environment is operating correctly and provide a baseline for any other RL controllers you develop.

closed 2018-07-09 by wil3

lengjia, 2018-07-09:
got it! thanks so much!

cbekar, 2019-06-19:
Lack of RL glue code makes reproducing the paper's results somewhat cumbersome.