wil3/gymfc #8: gazebo9

fangchen1993, 2018-07-08:
Hi,I have installed the gazebo9.There are some problem when i run the example  "python3 -m gymfc.controllers.iris_pid_eval --env-id=AttFC_GyroErr-MotorVel_M4_Ep-v0
",
I got error:
**Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/fc/gymfc/gymfc/controllers/iris_pid_eval.py", line 180, in <module>
    main(args.env_id, args.seed)
  File "/home/fc/gymfc/gymfc/controllers/iris_pid_eval.py", line 163, in main
    env = gym.make(env_id)
  File "/home/fc/gym/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/home/fc/gym/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/fc/gym/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/fc/gymfc/gymfc/envs/attitude_control_env.py", line 67, in __init__
    super(GyroErrorESCVelocityFeedbackEnv, self).__init__(motor_count = motor_count, world=world)
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 124, in __init__
    self.gz_port = self._get_open_port(self.GZ_START_PORT)
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 266, in _get_open_port
    open_ports.append(c.laddr.port)
AttributeError: 'tuple' object has no attribute 'port'
**

wil3, 2018-07-09:
Hi @fangchen1993, please refer to the README, only Gazebo 8 is supported and I have no timeline when 9 will be supported. I"m open to PR if you get it working in  9 though, thanks.

fangchen1993, 2018-07-13:
Hi,@wil3,Thank you for your reply.I have build the environment

fangchen1993, 2018-07-13:
@wil3 I'm sorry that i met a new problem;
`Exception occured connecting to Gazebo retrying .... [Errno 111] Connect call failed ('127.0.0.1', 11345)
Gazebo multi-robot simulator, version 8.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.109.128
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  154 (DRI2)
  Minor opcode of failed request:  3 (DRI2CreateDrawable)
  Resource id in failed request:  0x4800002
  Serial number of failed request:  29
  Current serial number in output stream:  30
Exception occured connecting to Gazebo retrying .... [Errno 111] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 111] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 111] Connect call failed ('127.0.0.1', 11345)
Exception occured connecting to Gazebo retrying .... [Errno 111] Connect call failed ('127.0.0.1', 11345)
Traceback (most recent call last):
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 287, in _connect
    self.manager = await pygazebo.connect((self.host, self.gz_port)) 
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/home/fc/gymfc/gymfc/pygazebo/pygazebo.py", line 562, in handle_connect
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/home/fc/gymfc/gymfc/pygazebo/pygazebo.py", line 252, in callback_impl
    future.result()  # check for exception
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 402, in sock_connect
    return (yield from fut)
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 432, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.0.1', 11345)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/fc/gymfc/gymfc/controllers/iris_pid_eval.py", line 180, in <module>
    main(args.env_id, args.seed)
  File "/home/fc/gymfc/gymfc/controllers/iris_pid_eval.py", line 163, in main
    env = gym.make(env_id)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/usr/local/lib/python3.5/dist-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/fc/gymfc/gymfc/envs/attitude_control_env.py", line 67, in __init__
    super(GyroErrorESCVelocityFeedbackEnv, self).__init__(motor_count = motor_count, world=world)
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 145, in __init__
    self._start_sim()
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 240, in _start_sim
    self.loop.run_until_complete(self._connect())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/home/fc/gymfc/gymfc/envs/gazebo_env.py", line 293, in _connect
    if not self.manager:
AttributeError: 'GyroErrorESCVelocityFeedbackEnv' object has no attribute 'manager'
`

wil3, 2018-08-30:
Just an update, in branch https://github.com/wil3/gymfc/tree/reset-optimization I'm removing the pygazebo dependencies and moving the implementation to the plugin which is much more dependable. This should make transitioning to Gazebo 9 more painless.

fangchen1993, 2018-09-03:
Hi, @wil3 ,Now i want to train a new quadcopter model and using RL algorithms to do some outdoor flight,where can I change the params of the quadcopter in the program,I have found the folder gymfc/envs/assets/gazebo/models for some model.sdf files,but i don't where can change the params of a new quadcopter such as the moment of inertia.Now i can run the simulation,how can i write the code to the pixhawk(i can find the inner loop code of px4)?Now ,i just can write the code of nerual network to replace the PID controller in the innerloop

wil3, 2018-09-03:
Hi @fangchen1993, yes the quadcopter models are defined in the sdf files. I suggest reading up on the SDF format http://sdformat.org/spec and creating models in Gazebo http://gazebosim.org/tutorials?tut=build_model. Anything outside the environment (e.g. running on pixhawk) is out of scope. 

In the future if you have a question please submit a new issue.

srussvoll, 2018-09-26:
Hi,
I am considering extending your framework to train a network to do pose control. On my distro it is time consuming to install Gazebo 8 since it, along with a dozen dependencies, need to be compiled from source (not in distros repositories). So I thought I might as well help update the codebase to work with Gazebo 9 instead. Do you have any estimate of when 'reset-optimization' will be ready enough to create a 'update_to_gazebo_9'-branch?

wil3, 2018-09-26:
Hi @srussvoll that would be a great addition. PR #11 was supposed to do this but I'm not sure what the state is at the moment, maybe @Kuppharish could chime in. It would probably be a good idea to create a separate issue on adding this support so we can talk about what is needed to integrate it into GymFC and stay consistent with being an OpenAI gym.

Yes installing Gazebo from source is a nightmare. We tried to get IT to install it on our university servers on CentOS and it was a failure. We ended up having to stand up on our Ubuntu servers. 

That branch is pretty much ready to be merged into master. I can do that today or tomorrow. At that point I'd just try master as is and see if it runs on gz9, no changes may be needed. If any changes do need to be made it would be to either the plugin or sdf files.

srussvoll, 2018-09-27:
I made a PR for master. Will update reset-optimization as well soon.

wil3, 2018-09-28:
See PR #16 for details on gz9 migration

weixingsun, 2019-01-17:
https://github.com/wil3/gymfc/tree/reset-optimization is no longer available, is gazebo9 come with trunk ?

kevin-y-ma, 2019-01-17:
I just change "SetupFile=/usr/share/gazebo-8/setup.sh" to "SetupFile=/usr/share/gazebo-9/setup.sh" in "configs/iris.config"  and i get an error : "[Err] [Plugin.hh:180] Failed to load plugin /home/myy/RL/gymfc/gymfc/envs/assets/gazebo/plugins/build/libQuadcopterWorldPlugin.so: libgazebo_sensors.so.8: cannot open shared object file: No such file or directory".

Is it because my gazebo`s version is 9.0?

wil3, 2019-01-17:
@weixingsun that branch was merged into master. This issue is still open because its still in the process of being migrated. I just recently upgraded to Ubuntu 18.04 and Gazebo 9 so will begin  more rigorous tests soon.

@kevin-y-ma the latest in master is a migration to Gazebo 9. The error you are seeing is most likely due to the plugin being compiled for Gazebo 8. This needs to be done during install but for the moment you can delete the build directory from this directory, https://github.com/wil3/gymfc/tree/master/gymfc/envs/assets/gazebo/plugins and then follow that README.md to rebuild the plugin. Let me know if that works.

kevin-y-ma, 2019-01-17:
It works, thanks.

wil3, 2019-01-17:
Cool let me know if any other issues come up not much testing has been done to master since the big shift in the configuration.

wil3, 2019-01-17:
I'm pretty sure something similar to this setup file can be used to dynamically build the plugin, 
https://github.com/openai/mujoco-py/blob/master/setup.py
Will look into this soon.

wil3, 2019-01-18:
Updates on master now build the plugin automatically if installing in normal mode. Tested on Ubuntu 18.04 and Gazebo 9.

closed 2019-01-18 by wil3