wil3/gymfc #9: Broken Quadcopter connection

Kuppharish, 2018-08-30:
Hi, I've installed the package in Ubuntu 16.04 with Gazebo8. Every thing is fine with rendering the drone. But when I try to give action to the drone it doesn't move and and gives the following error:

>>action=np.array([0.2,0.2,0.3,0.3])
>>env.step(action)
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [0/5]
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [1/5]
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [2/5]
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [3/5]
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [4/5]
[Wrn] [QuadcopterWorldPlugin.cpp:530] Broken Quadcopter connection, count [5/5]
[Wrn] [QuadcopterWorldPlugin.cpp:539] Broken Quadcopter connection, resetting motor control.

Can you please tell me what the problem could  be. Thank you

wil3, 2018-08-30:
Hi Harish, Two things, (1) Do you have the PID example working? and (2) Can you post a minimum working example I can test? That error signifies there is something wrong with the network connection between the python interface and the Gazebo plugin. More specifically it means it didn't receive the entire command. Also if you are only giving it a single command you may not notice it moving. Each step is only 0.001 seconds. Try putting it in a loop.

Kuppharish, 2018-08-30:
The PID example is running and when I added the env.render() command in the PID scripts then I was getting proper result. I got the error when I was running in the python command line after importing gym and gymfc. These are the following commands I gave:

import gymfc
import gym
env=gym.make(' model name')
env.render()
action=np.array([0.2,0.2,0.3,0.3])
env.step(action)

Hope you can find out the problem with this information.

zbenic, 2018-08-30:
Try adding delay of a couple of seconds after the render command .

Kuppharish, 2018-08-30:
@wil3 sorry for the silly question it worked fined when I ran it in loop. 
Is there a way to remove the ball joint of drone and world so that I can get the translational motion and also the feedback of its translations speeds. 
Thank you

wil3, 2018-08-30:
@Kuppharish Glad you got it working. Looking at your example make sure in the future you call `env.reset()` before taking any steps. This clears some bookkeeping variables, resets the simulator and samples a new target. 

As for your second question, the FDMPacket does include  linear velocity and position information, https://github.com/wil3/gymfc/blob/master/gymfc/envs/gazebo_env.py#L49 so you can access this through the observation variable in GazeboEnv. To remove the ball joint you need to create a new world file. Gazebo world and model files are in SDF format, you can find more information about them here, http://sdformat.org/spec. I don't plan on implementing this unless my research brings me in that direction but if you get it working submit a pull request would be a cool addition. Good luck!

closed 2018-08-30 by wil3

OsamaZafar12, 2018-11-06:
Same issue  Could u please help me how you solve that issu eMy code is 
import os 
import time
os.environ["OPENAI_LOG_FORMAT"] = "stdout,log,csv,tensorboard"
os.environ["OPENAI_LOGDIR"] = os.path.join(os.path.dirname(__file__), "../tensorboard")
import gym
import gymfc
from baselines import run 
env_id ='AttFC_GyroErr-MotorVel_M4_Ep-v0'
env = gym.make(env_id)
while True:
 env.reset 
 env.render()	
 run.main()

gym fc environment is stucked and not showing anything just quadcopter is In standby position 
[image]

> @wil3 sorry for the silly question it worked fined when I ran it in loop.
> Is there a way to remove the ball joint of drone and world so that I can get the translational motion and also the feedback of its translations speeds.
> Thank you

wil3, 2018-11-06:
Hi @OsamaZafar12 

> gym fc environment is stucked and not showing anything just quadcopter is In standby position

Of course it is, that's what you are telling it to do, reset then render. When pasting code in github make sure to use syntax [highlighting](https://guides.github.com/features/mastering-markdown/) otherwise the format will be messed up. Then use the 'Preview' to make sure it looks the way you want.

OsamaZafar12, 2018-11-07:
Actually the quad-copter is stuck in the space.It is not showing any action/maneuvers.I've been using following 
```#!/usr/bin/python
import os 
import time
os.environ["OPENAI_LOG_FORMAT"] = "stdout,log,csv,tensorboard"
os.environ["OPENAI_LOGDIR"] = os.path.join(os.path.dirname(__file__), "../tensorboard")
import gym
import gymfc
from baselines import run 

env_id ='AttFC_GyroErr-MotorVel_M4_Ep-v0'
env = gym.make(env_id)
env.render()

run.main()
```

wil3, 2018-11-07:
Each time step is 0.001 seconds. You aren't going to visually see any change after doing a single step. Think of `step` as the simulation clock, the simulation is only going to progress as you call this function. It's much more helpful to be logging your actions and rates and then producing graphs to verify behavior. 

> Actually the quad-copter is stuck in the space.It is not showing any action/maneuvers.

Just like your initial post it is doing exactly as you programmed, create and then render the environment. I'm not sure why you expect an action to occur without calling step. I suggest reading the tech report if you haven't already, review the [PID controller code ](https://github.com/wil3/gymfc/blob/master/gymfc/controllers/iris_pid_eval.py), the [Gazebo plugin](https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/QuadcopterWorldPlugin.cpp) if you are still confused.