wil3/gymfc #19: Simplest way to use OpenAI Baselines with GymFC

srussvoll, 2018-10-05:
I am developing some RL algorithms, and want to compare them to the OpenAI Baselines. As a starting point I want to get PPO2 to work as this was the most successful Baseline in your paper. Is there an easy way to do this?

What I have tried is creating a new Python project installing GymFC and Baselines as dependencies. Baselines has categorized the environments after how they need to interface with them. GymFC uses the "envs" category which is not supported directly by Baselines. So I changed this to "classic_control" which uses a vectorized interface. Training PPO2 using default parameters, "eprewmean" (episode reward mean) does not converge even after 10 million time steps. It fluctuates at around -530 from the beginning of the training until the end. Plotting "omega_actual' and "omega_target" using "gymfc.controllers.iris_pid_eval.plot_step_response" shows the PPO2 controller does not converge to the setpoint at all.

GymFC uses Gazebo 9 (ref. PR #16) which works successfully with the "iris_pid_eval" controller. Is this how GymFC is designed to be used with Baselines?

wil3, 2018-10-05:
I ended up using PPO1 actually, I think PPO2 was new at the time. I took one of the running examples like this one, https://github.com/openai/baselines/blob/master/baselines/ppo1/run_mujoco.py (which looking at it has changed considerably since I used it btw but you get the idea) and modded it. Since baselines doesnt know of the gymfc environments you must do `import gymfc` at the beginning of the file to register the environments. Then you can use the environment ID, for example AttFC_GyroErr-MotorVel_M4_Ep-v0 and your are good to go. 

I'm not sure what you mean by "envs" category. This was about 6 months ago so baselines may have changed since then.

srussvoll, 2018-10-05:
They have changed the code in Baselines since I started running PPO2 with GymFC. Now `run.py` defaults to building the environment like this:
```
env = make_vec_env(env_id, env_type, args.num_env or 1, seed, reward_scale=args.reward_scale)
if env_type == 'mujoco':
    env = VecNormalize(env)
```
So now Baselines works with GymFC without any modifications. However PPO2 still does not converge.

> I'm not sure what you mean by "envs" category. 

All environments have a category so that different categories can initialize things differently. The category is specified like this:
`entry_point='GymName.CategoryName:EnvironmentName',`

The code in my project that imports both GymFC and Baselines is something like this:
```
import gymfc
from baselines import run

run.main()
```
To run:
`python run.py --alg=ppo2 --env=AttFC_GyroErr-MotorVel_M4_Ep-v0 --num_timesteps=10000000`

Could you try this on your system running Gazebo 8 on the GymFC master branch and see if it converges? If it does converge, we know the problem is either the upgrade to Gazebo 9 or the reset-optimization branch changes. Else it might be more probable that the GymFC environment doesn't work out of the box with Baselines. I think it would be a good thing to make GymFC be directly compatible with Baselines as this is a good baseline to compare your own algorithms.

wil3, 2018-10-05:
Yea I can test that out. For the paper I forked baselines and instrumented logging to be able to generate plots for individual step responses and also to track progression of training.   How did you confirm its not converging? Is this based on doing a step response analysis on the trained network?

srussvoll, 2018-10-05:
Thank you for testing this.

>  How did you confirm its not converging?

By adding this to my `run.py` script:
```
os.environ["OPENAI_LOG_FORMAT"] = "stdout,log,csv,tensorboard"
os.environ["OPENAI_LOGDIR"] = os.path.join(os.path.dirname(__file__), "../tensorboard")
```
This makes Baselines log the training progress in a format readable by Tensorboard. This can then be viewed in the web browser: `tensorboard --logdir=tensorboard`

Looking at the eprewmean graph, it does not increase, rather it stays at around -530.

wil3, 2018-10-05:
I'm working on a deadline so I wont be able to get to it for a week or so, in the mean time can you plot the step response from the trained network and see how accurate it is? You can see signs of the step response forming just after ~100,000 steps. Its possible it converged very fast. If you look at the training plots in the paper this is what we observed.

wil3, 2018-10-05:
That says PID step response, did you write code to load the trained graph and evaluate the model?

srussvoll, 2018-10-05:
> did you write code to load the trained graph and evaluate the model?
Yes, by using the `--play` flag. I modified the baselines.run.main function to return desireds and actuals and plotted it using `gymfc.controllers.iris_pid_eval.plot_step_response`

wil3, 2018-10-05:
Yea somethings not right, I'm going to have to look into how the new baselines is working. Only other thing I can suggest for now is confirming the model that is being evaluate is indeed being updated through out training (or at least at the very end). None of this functionality existed when I was using it, I had to add all the checkpointing so I'm not familiar with what they are doing.

srussvoll, 2018-10-16:
Great, thank you for your comment. I'll leave a message when I have tested this.