wil3/gymfc #20: How can I modify the simulation time?

xianjiez, 2018-10-30:
Hi,I want to change the simulation time from 1s to 10s,how can I get it?

wil3, 2018-10-30:
https://github.com/wil3/gymfc/blob/master/gymfc/__init__.py#L10

closed 2018-10-30 by wil3

xianjiez, 2018-10-31:
Thank you for your answer and I have another doubt. I saw that there was jitter when the simulation reached the expected value, and what caused the jitter?In the simulation environment, whether a disturbance is added（e.g. wind）？

wil3, 2018-10-31:
Hi @xianjiez Do you mean from the plots in the paper or the PID example? There is no added disturbances, any observation you see in the step response is from the controller. Jitter is usually a sign of over correction. The jitter in the PID plot is extremely minor and won't be noticeable in flight. Neural networks on the other hand are more difficult to tame.