wil3/gymfc #23: Unclear what Gazebo `SetupFile` is referring to in the README

erniejunior, 2019-02-10:
While trying to get the GymFC up and running by running the iris pid example I ran into the following issue:
```bash
➜  controllers git:(master) python3 run_iris_pid.py
Loading config from  ../configs/iris.config
'tuple' object has no attribute 'port'
```
In the readme, you mention that the Gazebo `SetupFile` needs to be pointing to the correct location. As somebody who has never used Gazebo before it was unclear which file exactly you refer to and what I should put in there.

wil3, 2019-02-11:
Hi @erniejunior is that the entire output? Looks like stuff is missing. Will need more information to be able to be able to help. What does your iris_config file look like? What Python and OS versions are you running? What is the output of `pip3 freeze`?

The Gazebo setup.sh file needs to be sourced to load some environment variables. In the example config if you are using Gazebo 9 on Ubuntu that should be the correct location. You can navigate to the location to confirm, otherwise if it was placed somewhere different you just need to locate it and modify its location in the config file. The error you are getting isn't related though.

erniejunior, 2019-02-11:
Hi @wil3 thanks for the fast reaction. 
Yep this is all the output I get.
I did not touch the iris.config, so it is just the [same as on the master branch](https://github.com/wil3/gymfc/blob/master/examples/configs/iris.config)
I am using Linux Mint 18.2 Sonya which corresponds to Ubuntu Xenial I think. Python version is 3.5.2.
[pip3 freeze output is here](https://gist.github.com/erniejunior/4832dfbda0ee7d85fc5b30560edf9bf6)
I installed Gazebo 9 and the setup.sh referred in the iris.config exists and I can source it.

wil3, 2019-02-11:
My guess it has something to do with this [line](https://github.com/wil3/gymfc/blob/master/gymfc/envs/gazebo_env.py#L396)

Can you test this for me? From the terminal run `ipython3`. Then do,
```python
import psutil
psutil.net_connections()[0].laddr
```
Whats the output from the last command?

erniejunior, 2019-02-11:
The output is (was):
```python
In [1]: import psutil
In [2]: psutil.net_connections()[0].laddr
Out[2]: ('127.0.0.1', 56934)
```
so I figured I should upgrade psutil (I had version 3.4.2 installed, the current version is 5.5.0). After doing so:
```python
In [1]: import psutil
In [2]: psutil.net_connections()[0].laddr
Out[2]: addr(ip='::', port=22000)
```
Now the `run_iris_pid.py` script works as desired. Thanks for helping me fix my boring version problems (and sorry for bugging you with such minor issues). Maybe you could specify the minimum psutils version in the requirements?

closed 2019-02-11 by erniejunior

wil3, 2019-02-12:
Yes I definitely should, did you happen to review the API history to find out in what version that change was made?

erniejunior, 2019-02-13:
Seems to be in the API since 5.3.0 [see here](https://github.com/giampaolo/psutil/blob/master/HISTORY.rst#530) [and here](https://github.com/giampaolo/psutil/issues/928)