wil3/gymfc #24: I want to run OpenAi Gym for hectorquadrotor or Ardrone, I have Ros Kinetic + Gazebo 7 on Ubuntu16.04.

wajeehulhassanvii, 2019-03-27:
Can I setup this repository for working with Ros Kinetic + Gazebo 7 and hectorquadrotor?
Thanks

wil3, 2019-03-27:
Hi @wajeehulhassanvii GymFC hasn't been tested for 7, only 8 and 9. Version 2 currently in testing is aircraft agnostic so you can use whatever one you want, it doesn't care. As for Ros, what do you need it for?

wajeehulhassanvii, 2019-03-28:
Thanks for the reply @wil3. I wanted to train a drone using DDPG, I was wondering if I can get OpenAi to work with Ros Kinetic + Gazebo7 on Ubuntu 16.04? I got Ardrone to work on my current setup, just wasn't sure how can I get OpenAi to work with the current project.

wil3, 2019-03-28:
GymFC v0.1 is an OpenAI environment, the README.md has all the environment names documented on how to initialize it with `gym.make`. GymFC doesn't use Ros, just Gazebo. What are you planning to do with Ros?

wajeehulhassanvii, 2019-03-29:
I wanted to run drone simulator using ros, it could be Hectorquadrotor, Rotors or Ardrone?

wil3, 2019-03-29:
Ros is composed of a bunch of modules which can aid in robotic tasks, I was trying to understand what portions of Ros you need and if it makes sense for using GymFC. In either case Ros and Gazebo 7 aren't currently supported and there is no plan to support older versions of Gazebo. If you are able to get it working please submit a PR.

closed 2019-03-29 by wil3