wil3/gymfc #25: how to specificallly set Gazebo SetupFile pointing to the correct location?

fei123ilike, 2019-03-28:
Hi wil3,
I am trying to run this project on Ubuntu 16.04, how to properly set gazebo setup file, 
I tried export {} =  gymfc/gymfc/envs/assets/gazebo/models/quadcopter_attitude_control on terminal, and source the .bashrc file, but there is nothing happened. Can you help me?

wil3, 2019-03-28:
If you are trying to get the Iris example working you have to confirm the setup path in this config, https://github.com/wil3/gymfc/blob/master/examples/configs/iris.config#L9

fei123ilike, 2019-03-28:
Will this working to set with /usr/share/gazebo-7/setup.sh, because I installed ROS Kinetic with gazebo-7.

I encountered another problem, after I open gazebo in another terminal, I ran python3 run_iris_pid.py , then the error was:

Killing process with ID= 5392
Timeout, could not connect to Gazebo

wil3, 2019-03-28:
You are in new territory, there's not plan in the near future to go backwards in Gazebo version or support Ros so I'm not really sure if it will work. 

Could not connect to Gazebo error means the network connection between the python and the Gazebo plugin could not be established for some reason.

If you do get it to work please submit a pull request or post an update. 

I do suggest if you can to update everything and use the newest LTS of Ubuntu and Gazebo 9.

closed 2019-03-28 by wil3