wil3/gymfc #35: Cmake error when building GymFC

xabierolaz, 2020-02-21:
Dart 6.7.0 from source
Gazebo 10 installed following commands in docker as example [(https://github.com/MichaelManz/gymfc/blob/demo_docker/Dockerfile-demo#L48)]
Building GymFC. I'm getting some errors at this step

wil3, 2020-02-22:
Hey @xabierolaz Can you provide output as text and not a screenshot? Its difficult to read and cannot be searched.
It doesn't look like Gazebo was installed correctly. If you look in that CMakeLists.txt file its trying to find the gazebo package which it can't. 

Look at the value of your LD_LIBRARY_PATH environment variable and others, did you complete these steps? 
http://gazebosim.org/tutorials?tut=install_from_source&cat=install#LocalInstall

xabierolaz, 2020-02-24:
> Hey @xabierolaz Can you provide output as text and not a screenshot? Its difficult to read and cannot be searched.
> It doesn't look like Gazebo was installed correctly. If you look in that [...]

Did those steps too but still getting same erro when building ./build_plugin.sh

Here are full steps I followed and the error as text output:
`
DART FROM SOURCE

1  sudo apt install virtualbox-guest-utils virtualbox-guest-dkms
    2  sudo apt-get remove libdart*
    3  sudo apt-get install build-essential cmake pkg-config git
    4  sudo apt-get install libeigen3-dev libassimp-dev libccd-dev libfcl-dev libboost-regex-dev
    5  libboost-system-dev
    6  sudo apt-get install libopenscenegraph-dev
    7  apt-get update
    8  sudo apt-get update
    9  sudo apt-get install libopenscenegraph-dev
   10  git clone git://github.com/dartsim/dart.git
   11  cd dart
   12  git checkout tags/v6.7.0
   13  mkdir build
   14  cd build
   15  cmake ..
   16  make -j4
   17  sudo make install
   18  cd 

GAZEBO FROM SOURCE

   19  sudo apt-get remove '.*gazebo.*' '.*sdformat.*' '.*ignition-math.*' '.*ignition-msgs.*' '.*ignition-transport.*'
   20  sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
   21  wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
   22  sudo apt-get update
   23  wget https://bitbucket.org/osrf/release-tools/raw/default/jenkins-scripts/lib/dependencies_archive.sh -O /tmp/dependencies.sh
   24  GAZEBO_MAJOR_VERSION=10 ROS_DISTRO=dummy . /tmp/dependencies.sh
   25  echo $BASE_DEPENDENCIES $GAZEBO_BASE_DEPENDENCIES | tr -d '\\' | xargs sudo apt-get -y install
   26  hg clone https://bitbucket.org/osrf/gazebo /tmp/gazebo
   27  sudo apt install mercurial
   28  hg clone https://bitbucket.org/osrf/gazebo /tmp/gazebo
   29  cd /tmp/gazebo
   30  mkdir build
   31  cd build
   32  cmake ../
   33  apt search sdformat*
   34  apt search sdformat6*
   35  apt install sdformat6*
   36  sudo apt install sdformat6*
   37  cmake ../
   38  sudo apt install sdformat9*
   39  cmake ../
   40  sudo apt install sdformat*
   41  cmake ../
   42  sudo apt-get install libsdformat2-dev
   43  cmake ../
   44  sudo apt-get remove 'libsdformat*' 'sdformat*'
   45  sudo apt-get install build-essential libtinyxml-dev libboost-all-dev cmake mercurial pkg-config
   46  mkdir ~/sdf_source
   47  cd ~/sdf_source/
   48  hg clone https://bitbucket.org/osrf/sdformat
   49  cd sdformat
   50  hg up sdf_2.0
   51  mkdir build
   52  cd build
   53  cmake .. -DCMAKE_INSTALL_PREFIX=/usr
   54  sudo apt-get install doxygen
   55  cmake .. -DCMAKE_INSTALL_PREFIX=/usr
   56  make -j4
   57  make 
   58  cd/tmp/gazebo
   59  cd /tmp/gazebo
   60  ls
   61  cmake ../
   62  cd build
   63  cmake ../
   64  apt search sdformat9
   65  sudo apt install sdformat9*
   66  cmake ../
   67  sudo apt install sdf*
   68  cmake ../
   69  sudo apt isntall cmake*
   70  sudo apt install cmake*
   71  cmake ../
   72  apt-get install -y lsb-release gnupg2 software-properties-common     && apt-get remove '.*gazebo.*' '.*sdformat.*' '.*ignition-math.*' '.*ignition-msgs.*' '.*ignition-transport.*'
   73  sudo apt-get install -y lsb-release gnupg2 software-properties-common     && apt-get remove '.*gazebo.*' '.*sdformat.*' '.*ignition-math.*' '.*ignition-msgs.*' '.*ignition-transport.*'
   74  sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
   75  sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
   76  wget http://packages.osrfoundation.org/gazebo.key -O - | apt-key add - && apt-get update
   77  wget http://packages.osrfoundation.org/gazebo.key -O - 
   78  apt-key add - && apt-get update
   79  sudo apt-key add - && apt-get update
   80  apt-key add - 
   81  sudo apt-key add - 
   82  wget https://bitbucket.org/osrf/release-tools/raw/default/jenkins-scripts/lib/dependencies_archive.sh -O /tmp/dependencies.sh
   83  apt-get update
   84  sudo apt-get update
   85  /bin/bash -c "GAZEBO_MAJOR_VERSION=10 ROS_DISTRO=dummy source /tmp/dependencies.sh && echo \${BASE_DEPENDENCIES} \${GAZEBO_BASE_DEPENDENCIES} \${DART_DEPENDENCIES} | tr -d '\\' | xargs apt-get -y install"
   86  sudo /bin/bash -c "GAZEBO_MAJOR_VERSION=10 ROS_DISTRO=dummy source /tmp/dependencies.sh && echo \${BASE_DEPENDENCIES} \${GAZEBO_BASE_DEPENDENCIES} \${DART_DEPENDENCIES} | tr -d '\\' | xargs apt-get -y install"
   87  sudo  apt-get install -y mercurial libboost-all-dev
   88  hg clone https://bitbucket.org/osrf/gazebo /tmp/gazebo 
   89  cd ..
   90  hg clone https://bitbucket.org/osrf/gazebo /tmp/gazebo 
   91  hg up gazebo10_10.1.0
   92  mkdir build && cd build 
   93  ls
   94  rm -rf build
   95  ls
   96  mkdir build && cd build 
   97  cmake -DCMAKE_INSTALL_PREFIX=/home/$USER/local ../
   98  make -j4
   99  sudo make install
  100  echo "export LD_LIBRARY_PATH=<install_path>/lib:$LD_LIBRARY_PATH" >> ~/.bashrc
  101  echo "export PATH=<install_path>/bin:$PATH" >> ~/.bashrc
  102  echo "export PKG_CONFIG_PATH=<install_path>/lib/pkgconfig:$PKG_CONFIG_PATH" >> ~/.bashrc
  103  source ~/.bashrc
  
GIT CLONE GYM FC 

  104  cd 
  105  git clone https://github.com/wil3/gymfc.git
  106  ls
  107  cd gymfc
  108  ls
  109  cd .
  110  cd ..
  111  ls
  112  cd gymfc
  113  pip install -e .
  114  sudo apt install python-pip
  115  pip install -e .
  116  ls
  117  cd gymfc
  118  ls
  119  cd envs
  120  ls
  121  cd assets
  122  ls
  123  cd gazebo
  124  ls
  125  cd plugins
  126  ls
  127  ./build_plugin.sh
ERROR:`

wil3, 2020-02-25:
Before you execute build_plugin.sh what is the output of,
` echo $LD_LIBRARY_PATH`
What is the output of,
`ls install_path>/lib `

xabierolaz, 2020-02-25:
echo $LD_LIBRARY_PATH
outputs nothing:
`xabi@xabi-VirtualBox:~/.local/share/Trash/files/gazebo/build$  echo $LD_LIBRARY_PATH

`

wil3, 2020-02-25:
Yea something is definitely wrong with your install.

Your replacing <install_path> with your actual install path right? Your output after sourcing it not making it seem that way.

Also it's very odd to place the build in a Trash directory.

wil3, 2020-02-25:
It's also not clear above why your doing multiple gazebo installs, i.e., line 32 and 98. I recommend starting from stratch and following the Gazebo install from source guide.

xabierolaz, 2020-02-26:
**Got it working!** @wil3 

The install path wasn't right

`Prepare Gazebo and Dart build

sudo apt-get install -y lsb-release gnupg2 software-properties-common
sudo apt-get remove -y '.*gazebo.*' '.*sdformat.*' '.*ignition-math.*' '.*ignition-msgs.*' '.*ignition-transport.*'
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add - && sudo apt-get update
wget https://bitbucket.org/osrf/release-tools/raw/default/jenkins-scripts/lib/dependencies_archive.sh -O /tmp/dependencies.sh
sudo apt-get update

Install Gazebo dependencies

sudo /bin/bash -c "GAZEBO_MAJOR_VERSION=10 ROS_DISTRO=dummy source /tmp/dependencies.sh && echo \${BASE_DEPENDENCIES} \${GAZEBO_BASE_DEPENDENCIES} \${DART_DEPENDENCIES} | tr -d '\\' | xargs apt-get -y install"

# Build Dart

sudo apt-get install -y ssh gdb vim git 
git clone "https://github.com/dartsim/dart.git" /tmp/dart 
cd /tmp/dart && git checkout v6.7.0 
mkdir build && cd build 
cmake .. 
make -j4 
sudo make install 
cd
rm -rf /tmp/dart

# Build Gazebo

sudo apt-get install -y mercurial libboost-all-dev 
hg clone https://bitbucket.org/osrf/gazebo /tmp/gazebo 
cd /tmp/gazebo 
hg up gazebo10_10.1.0 
mkdir build && cd build  
cmake -DCMAKE_INSTALL_PREFIX=/home/$USER/local ../ 
make -j4 
sudo make install 

echo "export LD_LIBRARY_PATH=/home/$USER/local/lib:$LD_LIBRARY_PATH" >> ~/.bashrc
echo "export PATH=/home/$USER/local/bin:$PATH" >> ~/.bashrc
echo "export PKG_CONFIG_PATH=/home/$USER/local/lib/pkgconfig:$PKG_CONFIG_PATH" >> ~/.bashrc
source ~/.bashrc

rm -rf /tmp/gazebo
cd
git clone https://github.com/wil3/gymfc.git
sudo apt install python3-pip -y
pip3 install .
cd gymfc/gymfc/envs/assets/gazebo/plugins
./build_plugin.sh
`

closed 2020-03-11 by wil3