wil3/gymfc #38: test_start_sim.py wont find components and run

xabierolaz, 2020-03-30:
Hi, im dealing some issues with running test environments,
Whenever I try to run **test_start_sim.py**
`(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_start_sim.py ~/nf1/model.sdf`
I get this error:
`Default configuration file at /home/cuda/projects/gymfc/env/lib/python3.6/site-packages/gymfc/envs/../../gymfc.ini missing, aborting`

Im using **nf1** model from the **thesis-work** branch 
so I went to the source code, and searched for that line, it’s inside the **gymfc/envs/fc_env.py** file :
[image]

so I guess it’s looking for **gymfc.ini** and not finding it inside current dir

test_start_sim.py is in gymfc/tests/**test_start_sim.py**, so wont be able to find gymfc.ini as its one directory up in gymfc/**gymfc.ini**, so Ive tried moving test_start_sim.py to project’s root folder ‘gymfc’ and running from there:

`python3 test_start_sim.py ~/nf1/model.sdf`

but then I get this new error:

`Could not find Gazebo setup.sh file at '/home/$USER/local/share/gazebo-10/setup.sh'. Typo?`

So I went there and **setup.sh** is clearly there, as I had pointed before inside the **gymfc.ini** file:
[image]

Any hints? Must be something trivial but have not been able to find the issue so far

wil3, 2020-03-30:
> (env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_start_sim.py ~/nf1/model.sdf

The first issue is unless you moved the default location of the model this isn't the correct command. It should be,

`python3 test_start_sim.py ../modules/digitaltwins/nf1/model.sdf`

Looking at the error message you are seeing your install path of gymfc looks like it wasn't installed in development mode (i.e., pip3 install -e .). If you look in ` /home/cuda/projects/gymfc/env/lib/python3.6/site-packages/gymfc/`you will indeed not find gymfc.ini (wouldn't make sense to put a config file there). If you choose to install it this way that is fine but you need to tell gymfc where that ini file is.

There are a couple ways to specify the gymfc.ini file. The first,  as the code you linked suggests, is to set it to the environment variable, 

`export GYMFC_CONFIG=../gymfc.ini; python3 test_start_sim.py ../modules/digitaltwins/nf1/model.sdf`

The second way is to provide it as a command line argument when running the test,

`python3 test_start_sim.py ../modules/digitaltwins/nf1/model.sdf --gymfc-config ../gymfc.ini`

Hope that helps.

closed 2020-03-30 by wil3

xabierolaz, 2020-04-06:
Thanks for the response!
Tried reinstalling as developer and **got gazebo running** when running the test:
`python3 test_start_sim.py ../modules/digitaltwins/nf1/model.sdf --gymfc-config ../gymfc.ini`
However, when gazebo starts **there is no** Neuroflight1 **model placed**, I have to manually go to the Insert tab inside gazebo and drag it into the scene.

There are no errors in the shell, so I'm just redoing steps and trying different methods.

When I run another test like the axis one, gazebo takes forever and I finally get a timeout mensage in the shell
`Starting gzserver with process ID= 4043
Roll Left
-----------------------------------
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1586170610.0324695
time_lapse_hours       0.01642354150613149

Killing Gazebo process with ID= 4043
Killing Gazebo process with ID= 4044
`

wil3, 2020-04-07:
What does your World tab look like in Gazebo?

There's something wrong with your install, look at your model and plugin paths...its pointing to two different locations, you may not have uninstalled the other gymfc. Make sure you are using virtual environments to prevent clashes like this. 

In the future if you have a separate issue could you please open a new one up?

xabierolaz, 2020-04-08:
Everything seems to work now, it was all about the gymfc install, fixed now thanks!

wil3, 2020-04-08:
Awesome! Glad to hear!