wil3/gymfc #39: build_plugin script fails with Boost error

SwapnilPande, 2020-04-06:
I am attempting build the plugin using the build_plugin.sh script. I get a Boost error:
```
/usr/include/boost/math/constants/constants.hpp:343:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function 'static constexpr T boost::math::constants::detail::constant_root_two_div_pi<T>::get(const mpl_::int_<5>&)':
/usr/include/boost/math/constants/constants.hpp:344:3: error: unable to find numeric literal operator 'operator""Q'
```

I tried both of the suggested flags `-std=gnu++11` and `fext-numeric-literals`. Neither of the flags change the error received. I also tried `-DBOOST_MATH_DISABLE_FLOAT128` based on similar issues for other packages, but received the error:
```
c++: fatal error: no input files
compilation terminated.
/bin/sh: 1: -std=c++11: not found
```

I was able to follow all of the other steps in the installation process without any problems, including building DART v6,7 and Gazebo 10 from source.

Some parameters:
Running in Docker container
Ubuntu 16.04
Boost version 1.58
g++/gcc version 6.50

SwapnilPande, 2020-04-06:
After more debugging, I was able to get closer to compiling using clang-6.0. I am still getting one error in compilation:
```
/root/gymfc/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp:195:26: error: use of undeclared identifier 'TRUE'
  this->world->SetPaused(TRUE);
```

SwapnilPande, 2020-04-06:
I was able to get around the error by changing:
```
this->world->SetPaused(TRUE);
```
to 
```
this->world->SetPaused(true);
```

This fix seems hacky and wrong, I would still appreciate your input on what you think the better way to solve this compiler issue would be.

wil3, 2020-04-07:
Hi @SwapnilPande,
My guess is because your Boost version is old, I'm currently running Ubuntu 18.04 with Boost 1.65 gcc 7.4.0.
Sorry the install is currently a bit of a mess that we need help with. I'm working to get a Docker image built. Let me know if that fixes the issue.
Cheers,
Wil

SwapnilPande, 2020-04-08:
I'll try it later tonight and let you know how it goes. Don't worry about it being a mess, I get how the development process goes.

I would love to help develop this package and get v2 stable, if you're okay with that. I've been debugging a couple of errors that I've encountered as I try to get this to run.

wil3, 2020-04-08:
That would be a huge help, now that I'm done with my degree its hard to find time to maintain this in my free time. I'm in the process of adding the rest of my thesis work which contains the NF1 model and new reward functions here, https://github.com/wil3/gymfc/tree/thesis-work

There is also a PR here for a Docker container I'm just looking for some additional changes before it gets merged, https://github.com/wil3/gymfc/pull/26

SwapnilPande, 2020-04-08:
I tried replicating your environment, but I'm still getting the same error... Not really what the difference is now.

wil3, 2020-04-09:
Starting from an unmodified clone of the repo, can you provide me with the exact steps and output you get?

wil3, 2020-04-12:
Hi @SwapnilPande consider using this docker image or following the install instructions within it to get a correct environment set up, https://github.com/wil3/gymfc/blob/master/Dockerfile-demo. @MichaelManz and I have been testing this image and it does not produce the compiling errors you are referring too.

wil3, 2020-04-14:
@SwapnilPande I'm reviewing the code, I'm opening this back up because I noticed inconsistencies of true and TRUE being used in the code which should be fixed. I also reviewed the Gazebo source code which has instances of both being used. I still don't know why you get a build fail from it but it should be fixed. Please submit a PR for this change if can. Thanks,
Wil

wil3, 2020-04-14:
Could you provide the output for running cmake and make? If you are still getting the error there is still an underlying problem with your environment we should try and identify. You also mentioned you are running in a docker container, could you share so we can compare to the one in this repo? That should be able to identify where the problem is.

SwapnilPande, 2020-04-19:
Sorry, had a busy week. To your first comment, do you want me to submit a PR to replace `TRUE` with `true`? Do we want to debug why `TRUE` is causing the build error?

I was able to get everything running correctly using the following versions:
Ubuntu 18.04
Gazebo 10
Dart 6.9.2

SwapnilPande, 2020-04-19:
This is the output of cmake and make with `TRUE`
```
-- Found DART: /usr/local/include (Required is at least version "6.6") found components:  dart
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0")
-- Boost version: 1.65.1
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread;-lpthread (found version "3.0.0")
Gazebo version: 10.2
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/gymfc/gymfc/envs/assets/gazebo/plugins/build
[ 68%] Built target sensor_msgs
[ 81%] Built target control_msgs
Scanning dependencies of target FlightControllerPlugin
[ 86%] Building CXX object CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o
/tmp/gymfc/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp: In member function 'virtual void gazebo::FlightControllerPlugin::Load(gazebo::physics::WorldPtr, sdf::ElementPtr)':
/tmp/gymfc/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp:195:26: error: 'TRUE' was not declared in this scope
   this->world->SetPaused(TRUE);
                          ^~~~
CMakeFiles/FlightControllerPlugin.dir/build.make:62: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o' failed
make[2]: *** [CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/all' failed
make[1]: *** [CMakeFiles/FlightControllerPlugin.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

wil3, 2020-04-21:
Hi @SwapnilPande, yes we should replace https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L195 with 'true' since it is an inconsistency. Its the only case in the code using TRUE. It just so happens this will also fix your build as well.

And yes we should also find root cause to make sure this won't affect your development in the future as our docker container does not through this error. 

Is that the full output of your cmake? It should output the CXX compiler among other things. 

I did notice you are using a newer version of DART (6.9.2) than we have tested with (6.7). Have you tried this DART version?

My only guess at the moment is one of the header files is brining in a macro for TRUE in my environment and not yours.

SwapnilPande, 2020-04-21:
Sorry, I missed some lines when copying. Here is the full output.

cmake:
```
-- The C compiler identification is GNU 7.5.0                                                                                                      
-- The CXX compiler identification is GNU 7.5.0                                                                                                    
-- Check for working C compiler: /usr/bin/cc                                                                                                       
-- Check for working C compiler: /usr/bin/cc -- works                                                                                              
-- Detecting C compiler ABI info                                                                                                                   
-- Detecting C compiler ABI info - done                                                                                                            
-- Detecting C compile features                                                                                                                    
-- Detecting C compile features - done                                                                                                             
-- Check for working CXX compiler: /usr/bin/c++                                                                                                    
-- Check for working CXX compiler: /usr/bin/c++ -- works                                                                                           
-- Detecting CXX compiler ABI info                                                                                                                 
-- Detecting CXX compiler ABI info - done                                                                                                          
-- Detecting CXX compile features                                                                                                                  
-- Detecting CXX compile features - done                                                                                                           
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")                                                                                   
-- Checking for module 'bullet>=2.82'                                                                                                              
--   Found bullet, version 2.87                                                                                                                    
-- Found Simbody: /usr/include/simbody                                                                                                             
-- Found CCD: /usr/include (found version "2.0")                                                                                                   
-- Found FCL: /usr/include (found version "0.5.0")                                                                                                 
-- Found ASSIMP: /usr/include (found version "4.1.0")                                                                                              
-- Found DART: /usr/local/include (Required is at least version "6.6") found components:  dart                                                     
-- Looking for pthread.h                                                                                                                           
-- Looking for pthread.h - found                                                                                                                   
-- Looking for pthread_create                                                                                                                      
-- Looking for pthread_create - not found                                                                                                          
-- Looking for pthread_create in pthreads                                                                                                          
-- Looking for pthread_create in pthreads - not found                                                                                              
-- Looking for pthread_create in pthread                                                                                                           
-- Looking for pthread_create in pthread - found                                                                                                   
-- Found Threads: TRUE                                                                                                                             
-- Boost version: 1.65.1                                                                                                                           
-- Found the following Boost libraries:                                                                                                            
--   thread                                                                                                                                        
--   system                                                                                                                                        
--   filesystem                                                                                                                                    
[... cut ...]
   this->world->SetPaused(TRUE);
                          ^~~~
CMakeFiles/FlightControllerPlugin.dir/build.make:62: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o' failed
make[2]: *** [CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/all' failed
make[1]: *** [CMakeFiles/FlightControllerPlugin.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

SwapnilPande, 2020-04-21:
I actually originally tried with Dart version 6.7. However, I got build errors with Gazebo 10 when I use it. From my research into that issue, it appears that Gazebo 9+ introduced a fix for an issue that arose with Dart versions 6.9+, but that made it incompatible with older versions of Dart.

I am not sure why Dart 6.7 works for you but not for me. There is likely some root cause for all of these issues.

I have pushed my latest docker container with gymfc working to Docker Hub. You can pull it by running

```
docker pull swapnilpande/gymfc:latest
```

Hopefully this will help with debugging.

wil3, 2020-04-21:
Do you happen to know the errors you got?
I've been setting up Travis for CI and it appears Dart is missing dependencies in their instructions and I got some build errors initially. I believe they exist in the dependencies for Gazebo which is why it may have gone unnoticed. I've been following essentially the steps in docker images we have in this repo. Unfortunately the Travis build takes too long so I'm going to need to deploy the docker container to Travis to do the build.

SwapnilPande, 2020-04-23:
Going through the installation process again with a fresh ubuntu 18.04 docker container (can be pulled using `docker pull ubuntu:18.04`. DART v6.7 installs successfully

I get errors during installing gazebo:

First error I get is during cmake:
```
CMake Error at cmake/SearchForStuff.cmake:601 (find_package):
  By not providing "Findsdformat9.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "sdformat9", but CMake did not find one.

  Could not find a package configuration file provided by "sdformat9" with
  any of the following names:

    sdformat9Config.cmake
    sdformat9-config.cmake

  Add the installation prefix of "sdformat9" to CMAKE_PREFIX_PATH or set
  "sdformat9_DIR" to a directory containing one of the above files.  If
  "sdformat9" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:147 (include)
```

I resolved this issue by running `apt-get install libsdformat9-dev`

Next: I got cmake errors because of missing Ignition packages:

```
BUILD ERRORS: These must be resolved before compiling.
--      Missing: Ignition msgs5 library (libignition-msgs5-dev).
--      Missing: Ignition Transport (libignition-transport8-dev)
--      Missing: Ignition Common (libignition-common3-dev
--      Missing: Ignition Fuel Tools (libignition-fuel-tools4-dev
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:321 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.
```

Not sure why these dependencies aren't being installed with the other gazebo dependencies.

Once these dependencies are installed, I get an error during `make`:
```
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/local/include/dart/config.hpp:5,
                 from /usr/local/include/dart/dart.hpp:33,
                 from /tmp/gazebo/gazebo/physics/dart/dart_inc.h:24,
                 from /tmp/gazebo/gazebo/physics/dart/DARTCollision.hh:23,
                 from /tmp/gazebo/gazebo/physics/dart/DARTHeightmapShape.cc:20:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of 'void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<double, 3, 1>; Src = Eigen::Matrix<float, 3, 1>; Func = Eigen::internal::assign_op<double, float>]':
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:728:41:   required from 'Derived& Eigen::PlainObjectBase<Derived>::_set_noalias(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<float, 3, 1>; Derived = Eigen::Matrix<double, 3, 1>]'
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:537:19:   required from 'Eigen::PlainObjectBase<Derived>::PlainObjectBase(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<float, 3, 1>; Derived = Eigen::Matrix<double, 3, 1>]'
/usr/include/eigen3/Eigen/src/Core/Matrix.h:379:29:   required from 'Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::EigenBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<float, 3, 1>; _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]'
/tmp/gazebo/gazebo/physics/dart/DARTHeightmapShape.cc:68:63:   required from here
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:834:3: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
gazebo/physics/CMakeFiles/gazebo_physics.dir/build.make:1118: recipe for target 'gazebo/physics/CMakeFiles/gazebo_physics.dir/dart/DARTHeightmapShape.cc.o' failed
make[2]: *** [gazebo/physics/CMakeFiles/gazebo_physics.dir/dart/DARTHeightmapShape.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

This error has to do with Eigen datatypes. Have you seen this error>

wil3, 2020-04-25:
I have not seen those errors but there are a couple things I'm investigating for the build processes. Following the Gazebo install from source instructions, if you do a local install you must make sure to update your PATH environment variables, http://gazebosim.org/tutorials?tut=install_from_source&cat=install. If you do not then the cmake files will not be found. I only ran into this when building the GymFC plugins. I ported the demo docker file we have at the root of this repo to a [travis config](https://github.com/wil3/gymfc/blob/i50-travis/.travis.yml) that may help installing the environment, I'm also in the processes of converting these to docker files.

The other thing I noticed recently doing testing is if you install via `pip3 install .` the GymFC plugins do not appear to be being built. I've always installed in development mode i.e., `pip3 install -e .` so maybe this went unnoticed. Hoping to figure that out today.

SwapnilPande, 2020-05-02:
The last issue I'm seeing is actually the same as #52.

However, I just followed the steps exactly in the travis config file and I was able to get Gazebo to build successfully! 

I THINK the only difference between my installation and the installation steps in the travis config is the version of Gazebo. When switching to the gazebo 10 tag, I ran

```
hg up gazebo10
```

while your script runs
```
hg up gazebo10_10.1.0
```

There must be a breaking change between versions of Gazebo 10. If you'd like, I can submit a PR with a change to the README mentioning this detail. I can also add some tips that I found useful when I was running through the installation process.

wil3, 2020-05-07:
Nice! That would be concerning if a minor version can cause a break like that though. Could you do a diff or install the two side by side?

Yea that would be helpful to update the README thanks, we should pin specific versions because the environment does not appear very flexible.

wil3, 2020-05-30:
@SwapnilPande would this issue be closed with PR #60 ?

SwapnilPande, 2020-05-31:
Yes it will!

closed 2020-06-03 by wil3