wil3/gymfc #40: test_axis.py timeout

xabierolaz, 2020-04-08:
Whenever I run text_axis.py

`python3 test_start_sim.py ../modules/digitaltwins/nf1/model.sdf`

Gazebo opens with the nf1 in the middle, but shells wont get pass from here:

```
(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_axis.py ~/projects/gymfc/modules/digitaltwins/nf1/model.sdf
Sending motor control signals to port  9527
Gazebo Model Path = /home/cuda/local/share/gazebo-10/models::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/projects/gymfc/modules/digitaltwins
Gazebo Plugin Path = /home/cuda/local/lib/gazebo-10/plugins::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build
Starting gzserver with process ID= 3028
Roll Left
-----------------------------------
```

After 1 minute, this message appears and gazebo ends.
```
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1586362388.8515196
time_lapse_hours       0.016432642539342245

Killing Gazebo process with ID= 3028
Killing Gazebo process with ID= 3029
Timeout communicating with flight control plugin.
```
Watching the youtube video in your Channel, I see that after all these hyphens, this should appear, but wont get pass from there.
[image]

wil3, 2020-04-09:
99% of the time its because you don't have the plugins built and in the correct location. You get a time out because there are no motor plugins to talk to. The thesis branch is still WIP in progress so there is still stuff missing. 

Take a look here for instructions https://github.com/wil3/gymfc#directory-layout

So youll need to update the directory to be something like, /modules/digitaltwins/nf1/plugin/bulid/

If you are using these plugins https://github.com/wil3/gymfc-aircraft-plugins youll need to create soft links to the built IMU and motor model plugins (libgazebo_imu_plugin.so and libgazebo_motor_model.so).

wil3, 2020-04-09:
There's actually an open issue to make this more clear, https://github.com/wil3/gymfc/issues/37

SwapnilPande, 2020-04-09:
@xabierolaz What output do you get when you pass the `--verbose` tag to `test_start_sim.py`?

xabierolaz, 2020-04-10:
@SwapnilPande this is what I get passing the --verbose
(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_axis.py ~/projects/gymfc/modules/digitaltwins/nf1/model.sdf --verbose
Sending motor control signals to port  9722
Gazebo Model Path = /home/cuda/local/share/gazebo-10/models::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/projects/gymfc/modules/digitaltwins
Gazebo Plugin Path = /home/cuda/local/lib/gazebo-10/plugins::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build
Starting gzserver with process ID= 12609
Roll Left
-----------------------------------
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11375
[Msg] Publicized address: 172.18.83.252
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9722
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, /home/cuda/projects/gymfc/modules/digitaltwins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_imu_plugin.so: libgazebo_imu_plugin.so: cannot open shared object file: No such file or directory
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1586517499.5778537
time_lapse_hours       0.016418274839719138

Killing Gazebo process with ID= 12609
/bin/sh: 1: kill: No such process

Killing Gazebo process with ID= 12610
Timeout communicating with flight control plugin.

@wil3 thanks, will have a look at #37 and directory layout instructions

SwapnilPande, 2020-04-10:
Yeah, so as @wil3 mentioned earlier, it can't find the motor model plugin. I've created a PR (#42) with additional documentation on how to setup the aircraft model. Until it is merged, you can try following the instructions [here](https://github.com/SwapnilPande/gymfc/#setting-up-an-aircraft-model) in my fork of the repo.

wil3, 2020-04-12:
@xabierolaz did building and linking the aircraft-plugins resolve the error? Can this be closed?

xabierolaz, 2020-04-13:
Followed all steps building and linking the aircraft plugins and following the directory structure, and getting this now when running test_axis.py:
`(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_axis.py ../modules/digitaltwins/nf1/model.sdf --verbose
Sending motor control signals to port  9177
Gazebo Model Path = /home/cuda/local/share/gazebo-10/models::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/projects/gymfc/modules/digitaltwins
Gazebo Plugin Path = /home/cuda/local/lib/gazebo-10/plugins::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build
Starting gzserver with process ID= 27257
Roll Left
-----------------------------------
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11842
[Msg] Publicized address: 172.18.83.252
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9177
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, ../modules/digitaltwins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=1 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=2 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1586774205.728632
time_lapse_hours       0.016450484659936693

Killing Gazebo process with ID= 27257
Killing Gazebo process with ID= 27258
Timeout communicating with flight control plugin.
(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ 
`

wil3, 2020-04-13:
What's the output of,
`ls -l /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build`

Show me the build contents of the aircraft plugin repo.

xabierolaz, 2020-04-13:
```
### cuda@eim-alu-83252:~$ ls -l /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build
total 16
lrwxrwxrwx 1 cuda cuda 106 Apr 13 10:25 libcontrol_msgs.so -> /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/gymfc-aircraft-plugins/build/libcontrol_msgs.so
lrwxrwxrwx 1 cuda cuda 111 Apr 13 10:25 libgazebo_imu_plugin.so -> /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/gymfc-aircraft-plugins/build/libgazebo_imu_plugin.so
lrwxrwxrwx 1 cuda cuda 112 Apr 13 10:25 libgazebo_motor_model.so -> /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/gymfc-aircraft-plugins/build/libgazebo_motor_model.so
lrwxrwxrwx 1 cuda cuda 105 Apr 13 10:25 libsensor_msgs.so -> /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/gymfc-aircraft-plugins/build/libsensor_msgs.so
cuda@eim-alu-83252:~$ 
```

wil3, 2020-04-14:
Thanks @xabierolaz. Everything looks fine but you do have something things different than how I have it setup so could you try a couple things? I don't think it should cause problems but I've never cloned the gymfc-aircraft-plugin repo inside a digital twin. To rule out all possibilities could you move it to `~/projects`? I'm also not familiar with that method to create symbolic links, your `ls ` output ls looks correct though. 

Inside `~/projects/gymfc/modules/digitaltwins/nf1/plugins/build` could you first remove all files in the directory and then execute, 
```
ln -s ~/projects/gymfc-aircraft-plugins/build/libgazebo_imu_plugin.so libgazebo_imu_plugin.so
ln -s ~/projects/gymfc-aircraft-plugins/build/libgazebo_motor_model.so libgazebo_motor_model.so
```
My build directory only has these two files.

wil3, 2020-04-14:
Reviewing the code, this error occurs when the python environment can not communicate with the Gazebo environment over the UDP network. The flight control plugin will wait until it gets data back from the motors and IMU, if those plugins don't exist it waits indefinitely and times out.  This is usually the problem as I previously mentioned. If the above does not help it may require uncommenting some debug statements in the FlightControllerPlugin.cpp plugin to see if its possibly a networking issue. 

I did test out test_axis.py with the thesis branch and nf1 model and was able to run it without problems.

xabierolaz, 2020-04-14:
Went to a previous snapshot where gymfc was clean
1- Created /plugins/build directories inside nf1 folder
2- git cloned gymfc-aircraft-plugins this time inside projects folder
3- inside /projects/gymfc-aircraft-plugins:
mkdir build
cd build
cmake ../
make
4- went to build folder` /home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build/`
Folder is empty, but when I use this method to create symbolic links,

5- `cp --symbolic-link ~/projects/gymfc-aircraft-plugins/build/*.so ~/projects/gymfc/modules/digitaltwins/nf1/plugins/build`

BTW, this is the method @SwapnilPande mentioned in its repo https://github.com/SwapnilPande/gymfc/#setting-up-an-aircraft-model

when running this, 4 files are created there, so I'm guessing both **libcontrol_msgs.so** and **libsensor_msgs.so** shouldn't be there?
[image]

6- Deleted all 4 files
7-Run 
`cuda@eim-alu-83252:~/projects/gymfc/modules/digitaltwins/nf1/plugins/build$ ln -s ~/projects/gymfc-aircraft-plugins/build/libgazebo_imu_plugin.so libgazebo_imu_plugin.so`

`cuda@eim-alu-83252:~/projects/gymfc/modules/digitaltwins/nf1/plugins/build$ ln -s ~/projects/gymfc-aircraft-plugins/build/libgazebo_motor_model.so libgazebo_motor_model.so`
8-This time just the 2 files you mentioned are created, imu and motor
[image]

9-Proceed to run test_axis.py

```
(env) cuda@eim-alu-83252:~/projects/gymfc/tests$ python3 test_axis.py ../modules/digitaltwins/nf1/model.sdf --verbose
Sending motor control signals to port  9323
Gazebo Model Path = /home/cuda/local/share/gazebo-10/models::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/projects/gymfc/modules/digitaltwins
Gazebo Plugin Path = /home/cuda/local/lib/gazebo-10/plugins::/home/cuda/projects/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/projects/gymfc/modules/digitaltwins/nf1/plugins/build
Starting gzserver with process ID= 2603
Roll Left
-----------------------------------
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11853
[Msg] Publicized address: 172.18.83.252
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9323
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, ../modules/digitaltwins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=1 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=2 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1586855447.330785
time_lapse_hours       0.01643754972351922

Killing Gazebo process with ID= 2603
Killing Gazebo process with ID= 2604
Timeout communicating with flight control plugin.
```

wil3, 2020-04-14:
I'm not familiar with that window manager you are using, you mentioned using a VM, what OS are you running?

I suggest uncommenting the gzdbg statements in this function, rebuilding and providing the output, https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L593

Your problem isn't obvious, its going to take some deep diving into the source code to find out whats going on with your environment.

xabierolaz, 2020-04-15:
I'm using Ubuntu 18.04.3 LTS with the xfce desktop environment, thats why it looks that way 
[image]

It's indeed a VM with 16GB RAM, 4 cores and a Nvidia Quadro 6000 (it was built so it can use the gpu)

I will be uncommenting and writing the output here today

xabierolaz, 2020-04-16:
Re-building process:
1- went to a clean state with just gazebo and dart installed
2-Installed gymfc and noticed this error I didnt have noticed before when doing **pip3 install -e .**

<details><summary>error: invalid command 'bdist_wheel</summary>
<p>

(env) cuda@eim-alu-83252:~/projects/gymfc$ pip3 install -e .
Obtaining file:///home/cuda/projects/gymfc
Collecting gym (from gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/77/48/c43b8a72b916cc70896aa431b0fc00d1481ae34e28dc55e2144f4c77916b/gym-0.17.1.tar.gz (1.6MB)
    100% |################################| 1.6MB 614kB/s 
Collecting numpy (from gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/07/08/a549ba8b061005bb629b76adc000f3caaaf881028b963c2e18f811c6edc1/numpy-1.18.2-cp36-cp36m-manylinux1_x86_64.whl (20.2MB)
    100% |################################| 20.2MB 58kB/s 
Collecting protobuf (from gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/57/02/5432412c162989260fab61fa65e0a490c1872739eb91a659896e4d554b26/protobuf-3.11.3-cp36-cp36m-manylinux1_x86_64.whl (1.3MB)
    100% |################################| 1.3MB 883kB/s 
Collecting psutil>=5.3.0 (from gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/c4/b8/3512f0e93e0db23a71d82485ba256071ebef99b227351f0f5540f744af41/psutil-5.7.0.tar.gz (449kB)
    100% |################################| 450kB 2.4MB/s 
Collecting cloudpickle<1.4.0,>=1.2.0 (from gym->gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/ea/0b/189cd3c19faf362ff2df5f301456c6cf8571ef6684644cfdfdbff293825c/cloudpickle-1.3.0-py2.py3-none-any.whl
Collecting pyglet<=1.5.0,>=1.4.0 (from gym->gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/70/ca/20aee170afe6011e295e34b27ad7d7ccd795faba581dd3c6f7cec237f561/pyglet-1.5.0-py2.py3-none-any.whl (1.0MB)
    100% |################################| 1.0MB 1.1MB/s 
Collecting scipy (from gym->gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/dc/29/162476fd44203116e7980cfbd9352eef9db37c49445d1fec35509022f6aa/scipy-1.4.1-cp36-cp36m-manylinux1_x86_64.whl (26.1MB)
    100% |################################| 26.1MB 44kB/s 
Collecting six (from gym->gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/65/eb/1f97cb97bfc2390a276969c6fae16075da282f5058082d4cb10c6c5c1dba/six-1.14.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./env/lib/python3.6/site-packages (from protobuf->gymfc==0.1.2)
Collecting future (from pyglet<=1.5.0,>=1.4.0->gym->gymfc==0.1.2)
  Downloading https://files.pythonhosted.org/packages/45/0b/38b06fd9b92dc2b68d58b75f900e97884c45bedd2ff83203d933cf5851c9/future-0.18.2.tar.gz (829kB)
    100% |################################| 829kB 1.3MB/s 
Building wheels for collected packages: gym, psutil, future
  Running setup.py bdist_wheel for gym ... error
  Complete output from command /home/cuda/projects/gymfc/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-tvmb034u/gym/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmps_139zw1pip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for gym
  Running setup.py clean for gym
  Running setup.py bdist_wheel for psutil ... error
  Complete output from command /home/cuda/projects/gymfc/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-tvmb034u/psutil/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpasg4d_p5pip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for psutil
  Running setup.py clean for psutil
  Running setup.py bdist_wheel for future ... error
  Complete output from command /home/cuda/projects/gymfc/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-tvmb034u/future/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpwxvavi5ipip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
[... 293 lines cut ...]
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[... cut ...]
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1587024379.6890924
time_lapse_hours       0.01643631484773424

Killing Gazebo process with ID= 12207
Killing Gazebo process with ID= 12208
Timeout communicating with flight control plugin.
</p>
</details>

wil3, 2020-04-16:
If you aren't seeing the debug statements that you uncommented that could mean the packets sent from python are never getting to the plugin for some reason. You'll need to trace the plugin to see where it's getting too. If you aren't receiving any packets there's a networking problem somewhere on your setup. You can use Wireshark to capture the UDP packets being sent from the Python code. The log messages tell you which UDP port is used to you can filter the packet stream on that. If you need help making sense of the Wireshark output upload it here and I'll take a look. I've never tested it within a VM so I cant say if that's a source of the problem. 

Other option is to work off of the docker images that have been tested to work.

wil3, 2020-04-16:
Following your steps above you built the plugin and the made the code change. You made sure to rebuild the GymFC plugin after the code change right? (The build_plugin.sh script)

wil3, 2020-04-17:
@xabierolaz Sorry I've been on mobile, was able to read through your post in greater detail. 
You write, 

> cd ~/projects/gymfc/gymfc/envs/assets/gazebo/plugins
uncommented gzb lines you told me inside FlightControllerPlugin.cpp
.../build_plugin.sh
as it's uncommented, building throws some errors and cannot be completed, most of them about not declaring f variables
(im guessing you meant just to modify the .cpp and run test_axis.py, not rebuilding gymfc completely?)
In this case, I will build build_plugin.sh without modifying the FlightController.cpp

The purpose of this test was to trace in FlightController.cpp where you are getting so it can help you figure out what's going on in your setup. Since you didn't make any changes there's nothing new here to use to help debug. If uncommenting gzdbg statements cause error then put your own in that just print out where you are in the code. Most importantly you need to find out if you are getting [here](https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L749). If  you aren't then somehow your client and server are not communicating.

xabierolaz, 2020-04-17:
Okay, I was uncommenting the https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L593 loop, not the https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L749
so I will uncomment that too, rebuild it and copy the output here

wil3, 2020-04-17:
1. It should be. I've never scene that error before and gymfc doesn't use wheels (but its dependencies may). I can double check this weekend. 
2. Doesnt matter where virtual environments are installed, they can be anywhere. I typically name them env and put them in the root of projects (which is why its in .gitignore). 

As a sanity check I recommend printing out a debug statement in the plugin as soon as it starts to make sure you are using the new one you are modifying. Maybe like right after this statement, "[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9308". 

Wireshark can verify the python side is sending UDP packets, just filter the UDP port by the port specified in the logs, (i.e., 9308 in the example above). The init sets the hostname to localhost, also make sure thats being resolved. On your machine if you just ping localhost it'll do the translation.

xabierolaz, 2020-04-20:
Quick update, installing wheel is not needed and therefore there are no errors when installing gymfc with pip3 install -e . without using any virtual environment. I will look forward to this while I try Wireshark and commenting that line in the .cpp

xabierolaz, 2020-04-27:
Tried to run the docker, but my virtual machines docker engine  seems unable to run even a simple  hello word , 

`cuda@eim-alu-83252:~$ sudo docker run hello-world
docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "process_linux.go:449: container init caused \"rootfs_linux.go:58: mounting \\\"proc\\\" to rootfs \\\"/var/lib/docker/overlay2/e80f3eab4e46d1de8e110f0674df5b39b7aba8e6e647def2bc5455fd9d2dc671/merged\\\" at \\\"/proc\\\" caused \\\"permission denied\\\"\"": unknown.
ERRO[0000] error waiting for container: context canceled 
`
so It might start from scratch with a physic computer

closed 2020-04-27 by xabierolaz