wil3/gymfc #41: Error in loading attitude_control_training_rig model

SwapnilPande, 2020-04-08:
I am attempting to initialize the environment, but I am getting an issue with the attitude_control_training_rig.

I call `FlightControlEnv` to initialize the environment, and the `gzserver` launches and returns the following error:
```
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'hello_base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
```
Looking at the Gazebo source code, it appears that SetAnchor just doesn't have an implementation, even in the latest version. I was wondering how you got around this issue to load the model.

wil3, 2020-04-09:
Can you post your code and the command you are executing? 

Did you get your other issue resolved?

wil3, 2020-04-09:
If you look at the test scripts, you can pass in verbose=True which can help with debugging, https://github.com/wil3/gymfc/blob/master/tests/test_axis.py#L48

SwapnilPande, 2020-04-09:
These are actually the errors I get when I'm running the test code `test_axis.py` and I pass `verbose = True`. The messages are the gzserver debug messages.

I've temporarily moved past the other issue with the hacky fix I mentioned earlier in that issue thread. However, I think it'd be better to properly find the root cause of that issue. I'll post an update in that thread in a little

SwapnilPande, 2020-04-09:
Just to confirm the versions of everything:
Ubuntu 18.04
Boost 1.65
Gazebo 10
Dart v6.7

SwapnilPande, 2020-04-09:
Without the debug messages, I simply get a timeout, this may be related to #40

SwapnilPande, 2020-04-09:
With more debugging, I have found that this is not an error that is causing the simulation to timeout, so I am closing this issue.

closed 2020-04-09 by SwapnilPande