wil3/gymfc #47: Create a docker container to be used headless

wil3, 2020-04-12:
The current docker file, tested for OSX, supports the Gazebo front end. However to do this requires the Mesa library which takes a considerably long time to build. For training a controller there is no need to include these libraries. 

Create a base image for headless use and then derive a second image with a front end.

SwapnilPande, 2020-04-21:
As I mentioned in the other issue, I have already started building a headless container. If we can identify the root cause in that issue, I can build a Dockerfile for that configuration.

wil3, 2020-05-30:
With the dependency script looking good I'm going to update the current docker image used for Travis to use the same script so the environments are all the same.

wil3, 2020-05-30:
Related to Issue: #58