wil3/gymfc #52: Error in running make command while installing Gazebo

varunag18, 2020-05-01:
Trying to setup gymfc on my Ubuntu 18.04. Installed DART 6.7.0 correctly. 
Following the steps mentioned in [http://gazebosim.org/tutorials?tut=install_from_source&cat=install](url) to install Gazebo 10.2.0, but facing the below error.
[image]
Running the command `ls /etc/apt/sources.list.d/ | grep dart` returns:
`dartsim-ubuntu-ppa-bionic.list`
`dartsim-ubuntu-ppa-bionic.list.save`
So, i believe my DART has installed successfully. Sorry, Im new to ubuntu, may be making a silly error.
Thanks in advance.
Varun

wil3, 2020-05-02:
Hi Varun,
Nothing to be sorry about, setting this environment up isn't the easiest task. It might be helpful to reference the steps [here](https://github.com/wil3/gymfc/blob/i50-travis/.travs.yml) and [here](https://github.com/wil3/gymfc/blob/master/Dockerfile-demo). These scripts are known to work to build the environment. Make sure you start off at a clean state though.

varunag18, 2020-05-06:
I have installed Gymfc sucessfully using the links u mentioned. However, im nt sure if my installation is fully correct. When I am in the folder /gymfc and run the following code at the terminal:
`python`
`import gymfc`
I dont get any error.
However, when I navigate to a subfolder in my directory, say /gymfc/tests and again execute the above code, i get the `ModuleNotFoundError: No module named 'gymfc' ` error. 
There is some elementary mistake i am making, owing to my lack of knowledge of Ubuntu and python in general. 
A couple of more questions i would like to add, though im not sure if i should be asking them here or opening a separate issue for it. I am trying to test some reinforcement learning algos on UAVs, and for that i have installed the Open AI baselines package. 

1. When I run the below command in the /gymfc folder:
`python run_gymfc.py --env=AttFC_GyroErr-MotorVel_M4_Ep-v0`
I get the below error: 
`UnregisteredEnv: No registered env with id: AttFC_GyroErr-MotorVel_M4_Ep-v0`

[image]

2. Is there some Readme file or a help document that describes the process of coding to obtain Gymfc, baseline algo like DDPG and Gazebo? Not sure if im asking too much ;)
Thanks in advance.

wil3, 2020-05-07:
Hi @varunag18,
It almost appears as if you have gymfcv1 installed. The environment AttFC_GyroErr-MotorVel_M4_Ep-v0 existed in v1 and does not exist in v2 as gymfc has moved to a more generic architecture.  What is the file `run_gymfc.py` ?

If you aren't running gymfc in a virtual environment I recommend doing so and install it via pip in development mode (i.e., with the -e flag). The readme has the instructions for this. I've never run it via ipython so I'm not sure of the expected behavior. 

For my thesis I used [this](https://github.com/wil3/openai-baseline) fork of OpenAI baselines. If you haven't done so already check out my thesis for more information on the changes from v1 to v2. I have a [branch](https://github.com/wil3/gymfc/tree/thesis-work) I'm adding the rest of my thesis work to including the openai gym environments. If you have a look at the test scripts in /tests you'll see how environments are created.

Before trying to train make sure you can successfully run through all of the test scripts. If you do not have your own aircraft model you can use the NF1 model in my thesis branch for the time being.

varunag18, 2020-05-12:
Hi Wil, first of all, thnx fr ur prompt replies. I am still facing issues with the installation. In the 4th step, where i need to manually build the plugin running the script `gymfc/envs/assets/gazebo/plugins/build_plugin.sh` gives me the following error: `fatal error: gazebo/physics/dart/DARTModel.hh: No such file or directory` in the file `gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp`. 
[image]
Is it something like Gazebo not being able to locate Dart, though I have installed it.

wil3, 2020-05-13:
That error is saying it can't find the gazebo header. Usually this means there was a problem with your gazebo install. 

If you take a look at the gazebo install from source [doc](http://gazebosim.org/tutorials?tut=install_from_source&cat=install) there is a section titled "**Local Install**". Have you made sure your install has been added to the correct environment variables?

varunag18, 2020-05-13:
Hi Wil, my Gazebo is installed correctly. Running `gazebo` command from the terminal loads the software correctly. However, I still updated the paths in the bashrc file, you mentioned above. The error is still the same, as above.
Also, any thoughts on the Gazebo Dart error mentioned in the screenshot above.

xabierolaz, 2020-05-13:
Hi @varunag18, did you change the gymfc.ini file so it points to your gazebo path?

wil3, 2020-05-13:
Can you provide the full output of running `build_plugin.sh`?  Difficult to know the state of your system without this information.

varunag18, 2020-05-15:
> Can you provide the full output of running `build_plugin.sh`? Difficult to know the state of your system without this information.

I have removed my earlier installation of gymfc and done a fresh install using the [dockerfile](https://github.com/wil3/gymfc/blob/master/Dockerfile-demo). I run a test script to check if installation is correct and I receive `ModuleNotFoundError: No Module named 'gymfc' ` error. Attaching a screenshot below:
[image]

Also, I try to manually build the plugin running the script gymfc/envs/assets/gazebo/plugins/build_plugin.sh, I receive the same error as earlier. [Attached](https://pastebin.ubuntu.com/p/Y9HmBG77xy/) the output of the `build_plugin.sh`.
The `/usr/include/gazebo-10/gazebo/physics` directory does not have a `dart` folder.

varunag18, 2020-05-15:
> Hi @varunag18, did you change the gymfc.ini file so it points to your gazebo path?

The gymfc.ini file has the following content. 
`SetupFile = $HOME/local/share/gazebo/setup.sh`, which seems to me to be reflecting the correct location. Please correct me if I am wrong.
[image]

I have changed the SetupFile path in gymfc.ini from gazebo to gazebo-10 once, but no difference in result.

varunag18, 2020-05-17:
After many attempts at uninstalls and re-installs, iv managed to finally get gymfc installed successfully. I ran the test scripts in the tests subfolder and they are running. Thanks Wil and @xabierolaz for all the help. its really heartening when you are new to a field and there is some help around.

closed 2020-05-17 by varunag18

lookingforfeng, 2022-03-26:
> After many attempts at uninstalls and re-installs, iv managed to finally get gymfc installed successfully. I ran the test scripts in the tests subfolder and they are running. Thanks Wil and @xabierolaz for all the help. its really heartening when you are new to a field and there is some help around.
> 
> 在多次尝试卸载和重新安装之后，iv 终于成功安装了 gymfc。我在测试子文件夹中运行了测试脚本，它们正在运行。感谢 Wil 和@xabierolaz 的帮助。当你刚进入这个领域，有人帮助你的时候，这真的很振奋人心。

how did you  solve it