wil3/gymfc #53: Running dockerfile in 18.04, steps after building

xabierolaz, 2020-05-04:
I'm using a brand new computer with 18.04 installed (ubuntu, GNOME desktop)

Tested if my system is able to run dockers, and it seems so.
```
xabi@xabi:~$ sudo docker run hello-world

Hello from Docker!
This message shows that your installation appears to be working correctly.

To generate this message, Docker took the following steps:
 1. The Docker client contacted the Docker daemon.
 2. The Docker daemon pulled the "hello-world" image from the Docker Hub.
    (amd64)
 3. The Docker daemon created a new container from that image which runs the
    executable that produces the output you are currently reading.
 4. The Docker daemon streamed that output to the Docker client, which sent it
    to your terminal.

To try something more ambitious, you can run an Ubuntu container with:
 $ docker run -it ubuntu bash

Share images, automate workflows, and more with a free Docker ID:
 https://hub.docker.com/
For more examples and ideas, visit:
 https://docs.docker.com/get-started/
```
Now I downloaded the gymfc git and run dockerfile, after modifying the line:
ENV USER=**gymfc** to ENV USER=xabi (my current username, maybe not needed)

Some red lines appear, but overall docker gets built after finishing 23 steps
```
Step 23/23 : ENTRYPOINT ["/bin/bash", "-c"]
 ---> Running in 13ea96072021
Removing intermediate container 13ea96072021
 ---> 2d3428917169
Successfully built 2d3428917169
Successfully tagged gymfc:demo

```
How can I run the docker and test it with the test_axis or other scripts now? I'm not used to run dockers and dont know where to locate it.

Thanks!

wil3, 2020-05-07:
Nice! The docker stuff is still very experimental but how it currently is set up it allows you to run any command inside container. The README.md has one such example of how this can be done. I haven't been able to get the GUI working though which would make many of the test scripts how they are unusable put supposedly it works on osx.  A better use case would be to use the container to run a headless training session. 

If you have a clean copy of ubuntu I would suggest just following the commands in this [travis](https://github.com/wil3/gymfc/blob/i50-travis/.travis.yml) file or the docker file to set up the environment.

xabierolaz, 2020-05-07:
I've just followed the commands in the travis file from a clean copy of ubuntu.
What's the last step for?
`pip install . `

xabierolaz, 2020-05-08:
Okay, **test_axis** does work! Will try the other scripts

All the issues are regarding the test_start script, cant get it to work whatever system Im using, so it must be something wrong with that specific script. Even if I'm getting the dart setanchor error, still working

wil3, 2020-05-09:
Pretty sure the setanchor is normal behaviour. That surprises me test_axis works but test_start does not since test_axis essentially just extends test_start. What happens when you run the script? Are the gzserver and gzclient processes running (could check with `ps -aux | grep gzclient`)? What is the output when verbose flag is set to True?

There has been a recent change that does not exist in test_axis but is in test_start however. On [this](https://github.com/wil3/gymfc/blob/master/tests/test_start_sim.py#L31) line, can you remove `.wait()` and see if you observe different behavior? Make sure you have installed gymfc installed in development/editable mode (i.e., `-e` flag in `pip3 install -e .`) otherwise changes won't take affect. You can put a print statement before the line you change as a sanity check you have it installed the expected way.

xabierolaz, 2020-05-12:
Confirmed Gymfc was installed in development mode (pip3 install -e .)
test_axis  output:

closed 2020-05-12 by xabierolaz