wil3/gymfc #61: Gzserver error loading shard library libdart-collision-bulletd.so.6.7

varunag18, 2020-05-22:
After successful installation, when I tested the scripts in the /test folder using 
`python3 test_start_sim.py` or `python test_axis.py`, i got the following error: 
`gzserver: error while loading shared libraries: libdart-collision-bulletd.so.6.7: cannot open shared object file: No such file or directory`
On executing the same test scripts using `sudo python <filename>`, ran the scripts successfully by loading the twin in Gazebo. 
Running using `sudo` is a work around, I dont understand the reason for this error. Also, I could not locate this shared library `libdart-collision-bulletd` anywhere on the net. (Notice the 'd' after the bullet)

wil3, 2020-05-23:
I've never seen that error before but `No such file or directory` errors for .so files means the libraries cant be dynamically loaded which means for some reason your LD_LIBRARY_PATH is incorrect or the installation is different than expected. Needing sudo is another indicator something is wrong with the install as the sudo user will have different environment variables than your user account. 

Would need to knowdetails of your installation environment and the verbose output of test_start_sim.py to help more.

varunag18, 2020-05-23:
My bashrc file has
`export LD_LIBRARY_PATH=/home/varun/local/lib:/lib:/usr/lib:/usr/local/lib:/usr/local/lib/python3.6/dist-packages/gymfc/envs/assets/gazebo/plugins/build`

Below is the error when test_start_sim.py is executed:
>(fc) varun@varun:~/Documents/gymfc/tests$ python3 test_axis.py ../examples/gymfc_nf/twins/nf1/model.sdf --verbose
Sending motor control signals to port  9352
Gazebo Model Path = /home/varun/local/share/gazebo-10/models:/usr/local/share/gazebo-10/models::/home/varun/Documents/gymfc/gymfc/envs/assets/gazebo/models:/home/varun/Documents/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /home/varun/local/lib/gazebo-10/plugins:/usr/local/lib/gazebo-10/plugins::/home/varun/Documents/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/varun/Documents/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 6280
gzserver: error while loading shared libraries: libdart-collision-bulletd.so.6.7: cannot open shared object file: No such file or directory
Roll Left
-----------------------------------
>Timeout communicating with flight control plugin.

>Simulation Stats
-----------------
>steps                  0
packets_dropped        0
time_start_seconds     1590250360.8239877
time_lapse_hours       0.0164124290810691
/bin/sh: 1: kill: No such process
Killing Gazebo process with ID= 6280
/bin/sh: 1: kill: No such process
Killing Gazebo process with ID= 6281
Timeout communicating with flight control plugin.

Below is the execution of the same script using sudo

> (fc) varun@varun:~/Documents/gymfc/tests$ sudo python3 test_start_sim.py ../examples/gymfc_nf/twins/nf1/model.sdf --verbose
Sending motor control signals to port  9132
Gazebo Model Path = /home/varun/local/share/gazebo-10/models::/home/varun/Documents/gymfc/gymfc/envs/assets/gazebo/models:/home/varun/Documents/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /home/varun/local/lib/gazebo-10/plugins::/home/varun/Documents/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/varun/Documents/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 5647
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

>[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11534
[Msg] Publicized address: 172.17.0.1
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9132
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, ../examples/gymfc_nf/twins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=1 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=2 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
[Dbg] [gazebo_motor_model.cpp:263]  Motor 0 force=0 vel=0
[Dbg] [gazebo_motor_model.cpp:263]  Motor 1 force=0 vel=0
[Dbg] [gazebo_motor_model.cpp:263]  Motor 2 force=0 vel=0
[Dbg] [gazebo_motor_model.cpp:263]  Motor 3 force=0 vel=0
[Dbg] [gazebo_motor_model.cpp:263]  Motor 0 force=1.40523e-29 vel=3.87157e-12
[Dbg] [gazebo_motor_model.cpp:263]  Motor 1 force=1.40523e-29 vel=3.87157e-12
[Dbg] [gazebo_motor_model.cpp:263]  Motor 2 force=1.40523e-29 vel=3.87157e-12
[Dbg] [gazebo_motor_model.cpp:263]  Motor 3 force=1.40523e-29 vel=3.87157e-12

Please let me know if any more details are needed to find the cause of error.

wil3, 2020-05-24:
If you run `gazebo` at the command line without sudo does it give you the same error? 

That library, libdart-collision-bulletd that's not found is part of Dart which implies that dependencies is either not installed or not being linked to Gazebo. 

I searched for that file on my computer and it does not exist, I do however have `/usr/local/lib/libdart-collision-bullet.so.6.7` (bullet without the d). This [post](https://stackoverflow.com/questions/44140061/gzserver-error-while-loading-shared-libraries-libdart-so-6-1-cannot-open-shar) describes a similar problem which may help debugging. I would suggest uninstalling any instances of Dart and Gazebo and installing the environment using [this](https://github.com/wil3/gymfc/blob/i58-dep-script/install_dependencies.sh) script which should resolve your issues.

varunag18, 2020-05-24:
> If you run `gazebo` at the command line without sudo does it give you the same error?

Yes...it does.

> I searched for that file on my computer and it does not exist, I do however have `/usr/local/lib/libdart-collision-bullet.so.6.7` (bullet without the d). 

Exactly, I also checked and found the same.

> This [post](https://stackoverflow.com/questions/44140061/gzserver-error-while-loading-shared-libraries-libdart-so-6-1-cannot-open-shar) describes a similar problem which may help debugging. 

Had checked this stackoverflow page earlier, while trying to google this error. The ldd command simply informs that this libdart-collision-bulletd is missing.
> I would suggest uninstalling any instances of Dart and Gazebo and installing the environment using [this](https://github.com/wil3/gymfc/blob/i58-dep-script/install_dependencies.sh) script which should resolve your issues.

A silly question, I havent been able to find its answer on the net....how should I completely uninstall Dart and Gazebo. While installing, I had git cloned them in /tmp folder. There the build folder was created and make install executed. So, I dont understand the location from where `sudo make uninstall` command can be executed, as /tmp folder's contents autodelete. Below is the set of commands I have executed to remove dart and gazebo files.
`sudo apt-get remove --auto-remove dart*`
`sudo apt-get purge --auto-remove dart*`
`sudo apt-get remove --auto-remove libgazebo*`
`sudo apt-get remove --auto-remove gazebo*`
`sudo apt-get purge --auto-remove gazebo*`

I did a `sudo rm -r dart` at `usr/include/` and `usr/share/` where dart folder existed. 
In the folder `/usr/local/bin/` and `/home/varun/local/bin/` i deleted all files related to gazebo using 
`sudo rm gazebo gazebo-10.1.0 gz gz-10.1.0 gzclient gzclient-10.1.0 gzprop gzserver gzserver-10.1.0`
However, I still find many files related to gazebo in multiple folders like `/home/varun/local/` and `/usr/local/`. Should I go about manually deleting them too? 
There is one thing i recall, when I was running the `make install` command to install Dart and Gazebo, I had got some Permission Denied error, due to which I changed the command to `sudo make install`. Could that be the reason why my `python3 test_start_sim.py <model_path_name>` gives error but the command `sudo python3 test_start_sim.py <model_path_name>` works ??

varunag18, 2020-05-27:
Doing a fresh install on a new machine solved all issues.

closed 2020-05-27 by varunag18

wil3, 2020-05-29:
@varunag18 Glad you got it up and running! Sorry for the late response, uninstalling things when installed from source is a pain, but it sounds like you did the correct thing. Dart on their website says their make has an uninstall command but for gazebo they tell you to just manually delete the libraries.