wil3/gymfc #71: Failed to compile Gazebo plugin

blyspyder, 2020-06-04:
when i executive pip2 install .   I get an error:Failed to compile Gazebo plugin.  How can I solve it.
My system is ubuntu18.04

wil3, 2020-06-04:
Hi @blyspyder please read [CONTRIBUTING.md](https://github.com/wil3/gymfc/blob/master/CONTRIBUTING.md) before opening an issue and read the [README.md](https://github.com/wil3/gymfc) for installation instructions. 

Python2 is not supported.

closed 2020-06-04 by wil3

blyspyder, 2020-06-04:
> Hi @blyspyder please read [CONTRIBUTING.md](https://github.com/wil3/gymfc/blob/master/CONTRIBUTING.md) before opening an issue and read the [README.md](https://github.com/wil3/gymfc) for installation instructions.
> 
> Python2 is not supported.

THanke you for your reply. But I use the pip3 get the same error too.

wil3, 2020-06-04:
No one is able to help you if you don't provide the necessary information to reproduce your problem.
Please provide all the commands and all the output for installing the environment and GymFC.