wil3/gymfc #72: Install Failed

blyspyder, 2020-06-05:
when I run "pip3 install .", I got the error
[image]

varunag18, 2020-06-05:
There is another [issue](https://github.com/wil3/gymfc/issues/35#issue-568944812) logged for the same error. Plz check if it solves your problem.

varunag18, 2020-06-05:
Also, please avoid pasting screenshots of the errors you get. You should be copying the terminal output using ```.

blyspyder, 2020-06-05:
> Also, please avoid pasting screenshots of the errors you get. You should be copying the terminal output using ```.

Thank you very much. Iwill do this next time

blyspyder, 2020-06-05:
> There is another [issue](https://github.com/wil3/gymfc/issues/35#issue-568944812) logged for the same error. Plz check if it solves your problem.

I will try it immediately。

wil3, 2020-06-06:
To receive further help on this issue, please provide the output from the installation instructions listed on the readme, https://github.com/wil3/gymfc#quick-start

wil3, 2020-06-20:
Closing due to inactivity

closed 2020-06-20 by wil3