wil3/gymfc #73: How to train my model?

xultrarhino, 2020-06-05:
Hi Will, I am new to this business and do not quite understand how to compile and install all these programs correctly! I need your help! In the Neuroflight project it was written that first you need to create a model and train in GymFC! So you can explain how all this is done!

wil3, 2020-06-06:
Hi @xultrarhino please provide more information and ask specific questions. Review [CONTRIBUTING.md](https://github.com/wil3/gymfc/blob/master/CONTRIBUTING.md) and [README.md](https://github.com/wil3/gymfc/blob/master/README.md) for installation instructions and links to papers on specifics if you are looking for more information.

wil3, 2020-06-23:
Closing due to no follow up.

closed 2020-06-23 by wil3