wil3/gymfc #76: I want to add position controller feature to the code, i.e: knowing the current position of the quadcopter, we give it a target position and it is supposedly able to reach it..

Geo2001, 2020-06-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

wil3, 2020-06-17:
Hi @Geo2001 since it looks like you are new to GH I'm going to let this one slide. Providing a paragraph for a title and ignore the template is not OK. 

Maintaining this project takes time. Templates are in place to make this task easier. Normally if the template is ignored the issue would be automatically closed. Please fix if or will be closed. 

Based on the previous issue you opened (#74), the follow up issue was assumed to be an outline of the PR you intend to make. Please also familiarize your self with [CONTRIBUTING.md](https://github.com/wil3/gymfc/blob/master/CONTRIBUTING.md) if you have not already done so.

xabierolaz, 2020-06-19:
@Geo2001 I've opened a new issue with the feature you want to get implemented, as I've started working on it. Please have a look at #79 , and feel free to add anything.

wil3, 2020-06-23:
@Geo2001 closing this due to no followup. Feel free to contribute to the conversation in #79.

closed 2020-06-23 by wil3