wil3/gymfc #82: test_axis.py timeout

xabierolaz, 2020-07-16:
### Prerequisites
* [ X ] Put an X between the brackets on this line to verify you have done the following:
    * I have checked that your issue isn't already filed.
    * I am running Ubuntu 18.04
    * I installed the environment and GymFC according to these [instructions](https://github.com/wil3/gymfc/#quick-start). 
    
### Description
test_axis.py timeouts. GymFC and all aircraft plugins are built according to /examples

STEP 1 INSTALLATION
sudo MAKE_FLAGS=-j4 ./install_dependencies.sh
pip3 install -e .
gymfc/envs/assets/gazebo/plugins/build_plugin.sh

STEP 2 BUILDING AIRCRAFT PLUGINS in /examples

mkdir gymfc_nf/twins/nf1/plugins && cd gymfc_nf/twins/nf1/plugins
git clone https://github.com/wil3/gymfc-aircraft-plugins.git
mkdir build
cd build
cmake ../gymfc-aircraft-plugins
make

### Steps to Reproduce
with **--verbose**
python3 tests/test_axis.py --verbose examples/gymfc_nf/twins/nf1/model.sdf

```
(env) cuda@eim-alu-83252:~/workspace/gymfc$ python3 tests/test_axis.py --verbose  examples/gymfc_nf/twins/nf1/model.sdf
Sending motor control signals to port  9021
Gazebo Model Path = /usr/local/share/gazebo-10/models::/home/cuda/workspace/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /usr/local/lib/gazebo-10/plugins::/home/cuda/workspace/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 13694
Roll Left
-----------------------------------
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11404
[Msg] Publicized address: 172.18.83.252
[Wrn] [DARTPhysics.cc:96] Gravity vector is (0, 0, 0). Objects will float.
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9021
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, examples/gymfc_nf/twins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=1 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=2 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1594889248.8639956
time_lapse_hours       0.016416591140959

Killing Gazebo process with ID= 13694
Killing Gazebo process with ID= 13695
Timeout communicating with flight control plugin.

```
without **--verbose**
python3 tests/test_axis.py  examples/gymfc_nf/twins/nf1/model.sdf
```
(env) cuda@eim-alu-83252:~/workspace/gymfc$ python3 tests/test_axis.py examples/gymfc_nf/twins/nf1/model.sdf
Sending motor control signals to port  9917
Gazebo Model Path = /usr/local/share/gazebo-10/models::/home/cuda/workspace/gymfc/gymfc/envs/assets/gazebo/models:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /usr/local/lib/gazebo-10/plugins::/home/cuda/workspace/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 13171
Roll Left
-----------------------------------
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1594889048.2995117
time_lapse_hours       0.016415154271655612

Killing Gazebo process with ID= 13171
Killing Gazebo process with ID= 13172
Timeout communicating with flight control plugin.

```
3. See error

**Expected behavior**
It's expected to show  keyboard input to simulate steps 
[enter] = Step, n= NExt Command, q = Quit

**Additional context**
Have tried  too linking plugins manually inside build folder, as @varunag18 commented

```
 ln -s ~/workspace/gymfc/examples/gymfc_nf/twins/nf1/plugins/build/libgazebo_imu_plugin.so libgazebo_imu_plugin.so
 ln -s ~/workspace/gymfc/examples/gymfc_nf/twins/nf1/plugins/build/libgazebo_imu_plugin.so libgazebo_motor_model.so

```
Output
``
ln: failed to create symbolic link 'libgazebo_imu_plugin.so': File exists
``

So I check whats inside build folder
[image]

wil3, 2020-07-16:
Thanks for the detailed report. Do you observe the same behavior if you install GymFC without the -e flag?

xabierolaz, 2020-07-16:
Same output without the -e flag
```
(env) cuda@eim-alu-83252:~/workspace/gymfc$ python3 tests/test_axis.py examples/gymfc_nf/twins/nf1/model.sdf
Sending motor control signals to port  9279
Gazebo Model Path = /usr/local/share/gazebo-10/models::/home/cuda/workspace/gymfc/env/lib/python3.6/site-packages/gymfc/envs/assets/gazebo/models:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /usr/local/lib/gazebo-10/plugins::/home/cuda/workspace/gymfc/env/lib/python3.6/site-packages/gymfc/envs/assets/gazebo/plugins/build:/home/cuda/workspace/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 7736
Roll Left
-----------------------------------
Timeout communicating with flight control plugin.

Simulation Stats
-----------------
steps                  0
packets_dropped        0
time_start_seconds     1594904285.6580505
time_lapse_hours       0.016427347593837315

Killing Gazebo process with ID= 7736
Killing Gazebo process with ID= 7737
Timeout communicating with flight control plugin.
```

What would be the exact steps from having gymfc installed and build_plugin.sh built to actually running the test_axis.py script? I think I must be doing something wrong

wil3, 2020-07-16:
Your motor and IMU models are loaded it says it right in your output,

```
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=0 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=1 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=2 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_motor_model.cpp:204] Loading Motor number=3 Subscribed to /aircraft/command/motor
[Dbg] [gazebo_imu_plugin.cpp:55] Loading IMU sensor
```
Are you running this in a VM? I'll have to see if I can reproduce your steps.

xabierolaz, 2020-07-16:
I'm running it on a native Ubuntu, but I could try it on a VM

xabierolaz, 2020-07-17:
okay @wil3, I can confirm it's working on a VM but not on a native ubuntu system. Have reproduced exact steps above and the VM gets to the end of the test_axis.py script while the native one stays at Aircraft fixed to the world and eventually timeouts

wil3, 2020-07-17:
I suspect if you have it working in a clean VM there is a problem with your native install. All of your plugins are showing they are loaded and working. Its possible your network settings for some reason on your machine are preventing communication with Gazebo. Can you post the verbose output of the VM run here?

xabierolaz, 2020-07-19:
verbose output of the VM run
```
ython3 tests/test_axis.py --verbose examples/gymfc_nf/twins/nf1/model.sdf

Sending motor control signals to port  9271
Gazebo Model Path = /usr/local/share/gazebo-10/models::/home/xabi/projects/gymfc/gymfc/envs/assets/gazebo/models:/home/xabi/projects/gymfc/examples/gymfc_nf/twins
Gazebo Plugin Path = /usr/local/lib/gazebo-10/plugins::/home/xabi/projects/gymfc/gymfc/envs/assets/gazebo/plugins/build:/home/xabi/projects/gymfc/examples/gymfc_nf/twins/nf1/plugins/build
Starting gzserver with process ID= 2807
Roll Left
-----------------------------------
Gazebo multi-robot simulator, version 10.1.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11613
[Msg] Publicized address: 10.0.2.15
[Wrn] [DARTPhysics.cc:96] Gravity vector is (0, 0, 0). Objects will float.
[Dbg] [DARTModel.cc:72] Initializing DART model attitude_control_training_rig
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'pivot' and joint 'base_joint'.
[Err] [DARTJoint.cc:195] DARTJoint: SetAnchor is not implemented.
[Dbg] [FlightControllerPlugin.cpp:225] Binding on port 9271
[Dbg] [FlightControllerPlugin.cpp:408] CoT link=battery
[Dbg] [FlightControllerPlugin.cpp:410] Got COT from plugin 0 0 0.058
[Dbg] [FlightControllerPlugin.cpp:413] Num motors 4
[Dbg] [FlightControllerPlugin.cpp:443] Inserting digital twin from SDF, examples/gymfc_nf/twins/nf1/model.sdf.
[Dbg] [DARTModel.cc:72] Initializing DART model nf1
[Dbg] [DARTModel.cc:128] Building DART BodyNode for link 'frame' with a free joint.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'battery' and joint 'battery_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_1' and joint 'motor_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_2' and joint 'motor_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_3' and joint 'motor_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'motor_4' and joint 'motor_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_1' and joint 'prop_1_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_2' and joint 'prop_2_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_3' and joint 'prop_3_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'prop_4' and joint 'prop_4_joint'.
[Dbg] [DARTModel.cc:166] Building DART BodyNode for link 'fc_stack' and joint 'fc_stack_joint'.
[Dbg] [FlightControllerPlugin.cpp:527] Setting link to center battery
[Dbg] [FlightControllerPlugin.cpp:542] Aircraft model fixed to world
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[... cut ...]
[enter] = Step, n = Next Command, q = Quit[Dbg] [gazebo_motor_model.cpp:263]  Motor 0 force=9.91648e-10 vel=0.0325232
[Dbg] [gazebo_motor_model.cpp:263]  Motor 1 force=0.000130942 vel=11.8183
[Dbg] [gazebo_motor_model.cpp:263]  Motor 2 force=4.03853e-06 vel=2.07551
[Dbg] [gazebo_motor_model.cpp:263]  Motor 3 force=3.78908e-06 vel=-2.0104
Roll=-0.0000 Pitch=0.0001 Yaw=-1.8641
X=0.0000 Y=0.0000 Z=0.0000
X=1.0000 Y=-0.0000 Z=0.0000 W=-0.0000
M1=0.3106 M2=112.8560 M3=19.8197 M4=19.1979
[enter] = Step, n = Next Command, q = Quit

```

wil3, 2020-07-28:
Any updates on this? I suspect it will take opening up wireshark and making sure messages are being passed to the Gazebo flight controller plugin.

xabierolaz, 2020-07-28:
I will update this on Friday as cannot access the virtual machine atm

closed 2020-07-31 by xabierolaz