wil3/gymfc #83: E: Unable to locate package libignition-fuel-tools-dev

basilyohannan, 2020-07-17:


basilyohannan, 2020-07-17:
sudo apt-get install libignition-fuel-tools-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libignition-fuel-tools-dev

basilyohannan, 2020-07-17:
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
--2020-07-17 21:54:39--  http://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... failed: Name or service not known.
wget: unable to resolve host address ‘packages.osrfoundation.org’
gpg: no valid OpenPGP data found.

wil3, 2020-07-17:
Please use the bug issue if you suspect a bug. Opening an issue with no context or details is not helpful. 

This project and its dependencies do not use Mercurial. Make sure you have an up-to-date copy of the master branch and follow the instructions in the readme.

closed 2020-07-17 by wil3