wil3/gymfc #84: Calculate copter characteristics from blackbox logs [Improvement]

phgilde, 2020-08-05:
Ive read in the paper that every copter has to be modeled for gazebo. Wouldn't it be possible to calculate the characteristics (inertia, motor thrust curve etc.) of an invidual quadcopter from (betaflight) blackbox logs? For example the thrust curve could be calculate if the pilot does a slow punchout by mapping the acceleration against the motor speeds. 
This would make it easier to pretrain a model for a specific copter before letting the model fly the copter.

xabierolaz, 2020-08-06:
From an ideal point I think it would be useful, but I think a flight log reflects how the model reacted in a specific environment (wind, pressure) and therefore wouldnt be accurate for building it into gazebo, where the conditions are ideal (no wind) unless you want it to.

wil3, 2020-08-19:
Hi @phgilde estimating thrust and torque accurately from blackbox logs would be difficult. In the past I've come across some research in this space but this is out of scope for gymfc. I'd certainly be interested in seeing this explored though, if you end up working on this I'd be happy to link to your paper or project on the README as methods for model development.  As long you are able to derive the model parameters you can use different methods, gymfc doesn't enforce one over the other. 

Something that would be in scope for gymfc that would helpful is researching and integrating domain randomization. This paper https://arxiv.org/abs/1903.04628 explores parameters that would be helpful for gymfc to create more generalized models. If its something you'd like to contribute and work on, open a separate issue and we can discuss how this could be done in more detail.

closed 2020-08-19 by wil3