wil3/gymfc #90: PID tunning with pid_example.

JeongsooHa, 2020-09-27:
Hi, I am very interested in tunning PID gain these days.

What I want to do is to find optimal PID gain value, so that I can use pid_example without RL(PPO).
I want to get a pid_example result similar to that of PPO.

I have used the default PID(from code, https://github.com/wil3/gymfc/blob/master/examples/pid_example.py#L28), but it was not what I expected result. 
There was a lot of oscillation for yaw like the image below.
[image]
And also I have found a critical problem with the original PID, when the target angular velocity is set as a negative value(ex, -30, -40, -20).
(As far as I know, NF1 has a maximum angular velocity of 100deg/s and a minimum angular velocity of 100deg/s for a single task.)
[image]
I have used PID gain like this.
    r_pid = [2.4, 33.24, 0.033]
    p_pid = [4.2, 64.33, 0.059]
    y_pid = [2, 5, 0.0]

I have already tried to use PID gain from paper, but I know that PID gain from the paper is for Betaflight, not NF1.
Do you have any recommendations to tune PID gain with NF1?

wil3, 2020-09-29:
Hi @JeongsooHa what goal are you trying to accomplish? NF1 is the name of the quadcopter used during development, Neuroflight (NF) is the neuro-controller based fork of Betaflight.  

The PID gains used in my research were roughly based on Ziegler Nichols which is a popular traditional tuning method. The purpose of PID included in gymfc is primarily two fold: 1) verify installation of GymFC is setup correctly, 2) provide a baseline benchmark for other controllers. I've never validated the PID gains in the real world, or sought to find optimal ones. 

If your goal is to find optimal PID gains then you need to first figure out what optimization strategy you are going to use. If your goal is to use the derived gains in the real world l world suggestion first creating an interface to the SITL firmware (e.g., PX4 or Betaflight) or porting the PID algorithm used by the firmware to python. Its been on my list to do this for PX4 just haven't got around to it.

Let me know and I can help guide you in the right direction.

JeongsooHa, 2020-10-06:
Thanks for your response.
What I've understood is that you have never tried to find optimal PID gain for NF1 in the simulation.
Then, what is the meaning of the PID gain in the code? Is it a random number?
(at [https://github.com/wil3/gymfc/blob/master/examples/pid_example.py#L28])

Actually, I am trying to find optimal PID gain for NF1 with Ziegler Nichols, in the simulation, not the real world.
And I wonder how I can apply other models (not NF1) to GymFC.
Thank you!

wil3, 2020-10-06:
The comment right above the line you linked too should answer that question, [Ziegler Nichols](https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method) does not guarantee optimal gains, far from it in the case of quadcopter control. 

If you want to find optimal gains you need to use an optimization algorithm based on some cost function as I previously suggested. If you just want to do Ziegler Nichols, gymfc already has a helper function to get those gains [here](https://github.com/wil3/gymfc/blob/master/examples/gymfc_nf/controllers/ZieglerNichols.py). If you are trying to do Ziegler Nichols automatically, then you just need to script the algorithm to find the ultimate period/gain then use ZieglerNichols.py to get the gains.

closed 2020-10-08 by wil3

JeongsooHa, 2020-10-14:
Thanks for your response :)