wil3/gymfc #92: AWS-robomaker compatibility

mbilalai, 2020-09-29:
Hi.
I am keen to know if some sort of work, being done on integration of **gymFC** and gazebo client into an **AWS-robomaker** env.
I want to train the agent using cloud compute from AWS, just because of better **GPU** options, which will decrease the training time considerably.
Thank you.

wil3, 2020-09-29:
Hi @mbilalai, I've never used aws-robomaker before but if you can run Ubuntu 18.04 and have sudo permissions to install whatever software you need it should work. Right now Ubuntu 18.04 and Docker are the only supported running environments. If you get this working please consider making a PR to update the README.md or if you make a tutorial on how to set it up I can link to it from the README.md.

mbilalai, 2020-09-29:
Okay @wil3.
Thanks for the quick response.
I'll definitely make a tutorial or a PR if i get it to work.