wil3/gymfc #96: Protobuf exception when running tests/test_step_sim.py

morrisirrom, 2021-03-09:
### Prerequisites
* [ X ] Put an X between the brackets on this line to verify you have done the following:
    * I have checked that your issue isn't already filed.
    * I am running Ubuntu 18.04
    * I installed the environment and GymFC according to these [instructions](https://github.com/wil3/gymfc/#quick-start). 
    
### Description
I installed gymfc on a fresh 18.04 install, and followed the installation instructions.
However, when I try to run 
`python3 tests/test_step_sim.py --verbose examples/gymfc_nf/twins/nf1/model.sdf 1`

I get a number of errors such as
```
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
```

and then eventually I get this exception:
```
[libprotobuf FATAL ./google/protobuf/repeated_field.h:1078] CHECK failed: (index) < (current_size_): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_): 
```

### Steps to Reproduce
0. Install a fresh copy of gymfc on an 18.04 machine

1. Execute command python3 tests/test_step_sim.py --verbose examples/gymfc_nf/twins/nf1/model.sdf 1
```
...
[Dbg] [DARTLink.cc:482] DARTLink::SetAngularVel() doesn't make sense if the parent joint is not free joint (6-dof).
[Dbg] [DARTLink.cc:411] DARTLink::SetLinearVel() doesn't make sense if the parent joint is not free joint (6-dof).
[libprotobuf FATAL ./google/protobuf/repeated_field.h:1078] CHECK failed: (index) < (current_size_): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_): 
```
3. See error

**Expected behavior**
Mostly I would expect no exceptions--perhaps this is a benign problem?
I actually haven't successfully run this before, but I was expecting to see some value being applied to the motors, and some general movement of the drone.

wil3, 2021-03-09:
Hi @morrisirrom I really appreciate you providing all of the details to your issue. 

You are getting that error because you are only passing a single motor value 1 at the command line. The arguments defined [here](https://github.com/wil3/gymfc/blob/master/tests/test_step_sim.py#L43) specify the motor value must be provided for each motor/actuator of the aircraft. Since NF1 is a quadcopter you should run,

```
python3 tests/test_step_sim.py --verbose examples/gymfc_nf/twins/nf1/model.sdf 1 1 1 1
```
Which would be 100% throttle to each motor. The protobuf error you are seeing is basically saying 'hey I expect to see more values'. It would be great to have better error handling but its tricky since test_step_sim would need to know how many motors the aircraft has. You could do this by parsing the model.sdf and look for the number of the motor plugins and throw an error if the input doesn't match. 

That should fix the issue you are seeing but let me know if that's not the case.

closed 2021-03-09 by wil3

morrisirrom, 2021-03-09:
That makes a bunch of sense--Thank you so much for your response!