wil3/gymfc #97: help with running pid example

MKamyab1991, 2021-04-27:
Hi thanks for sharing.
I was unable to run the pid example, I have installed everything as you instructed however it gives the following error, meanwhile I am unaware of how an SDF file works. I am just curious about the way a pid is implemented on a quadcopter. I would appreciate it if you could help me run it. I have found 2 sdf files in all your repository, I put them both in the mentioned path examples/gymfc_nf/twins/nf1/but it still does not work out.
Thanks
```
/home/adrian/anaconda3/envs/attitude/bin/python /mnt/d/PycharmProjects/attitude_control_gymfc_will/examples/pid_example.py
2021-04-27 10:16:55.329255: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory
2021-04-27 10:16:55.329346: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
/mnt/d/PycharmProjects/attitude_control_gymfc_will/examples/pid_example.py
/home/adrian/anaconda3/envs/attitude/lib/python3.8/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
Aircraft SDF file  at location 'examples/gymfc_nf/twins/nf1/model.sdf' does not exist.
```

wil3, 2021-04-27:
Hi @MKamyab1991 please have a look at this tutorial, https://youtu.be/r6Q0Q7z2K9c.

I'd be better able to help debug if you ran it as outlined in the video and README but this looks like an issue with your environment. You can optionally try passing the full path to the SDF using the --twin flag.

SDF files are specific to Gazebo and describe the sim environment and models. Gazebo offers a number of tutorials online I recommend reading.

MKamyab1991, 2021-04-28:
Hi Will thanks a lot for your response
This is the error that I will get, I installed it in development mode so could you please help me how to manually build the plugins? 
Bests
```
python3 test_start_sim.py ../examples/gymfc_nf/twins/nf1/model.sdf
Sending motor control signals to port  9787
Could not find Gazebo plugins. If you have installed in development mode these must be built manually. Please refer to /mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/README.md for manually building the plugins.
```

wil3, 2021-04-28:
The output tells you what to do...the instruction are in that readme, did you do those steps?

MKamyab1991, 2021-04-29:
Thanks again. Yes I did. when I cmake ../ I get the following error. 
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Simbody_LIBRARIES (ADVANCED)
    linked by target "AircraftConfigPlugin" in directory /mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins
    linked by target "FlightControllerPlugin" in directory /mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins

-- Configuring incomplete, errors occurred!
See also "/mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/build/CMakeFiles/CMakeOutput.log".
See also "/mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/build/CMakeFiles/CMakeError.log".
```
the full log: 
`attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/build   master ±  cmake ../
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/simbody/SimbodyConfig.cmake:131 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:160 (find_package)
  CMakeLists.txt:4 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found ccd: /usr/include (found suitable version "2.0", minimum required is "2.0") 
-- Found fcl: /usr/include (found suitable version "0.5.0", minimum required is "0.3.2") 
-- Found assimp: /usr/include (found version "5.0.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
[... 28 lines cut ...]
-- Found JSONCPP: TRUE  
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Found YAML: TRUE  
-- Checking for module 'libzip'
--   Found libzip, version 1.5.1
-- Found ZIP: TRUE  
-- Looking for ignition-common3 -- found version 3.12.0
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.6.1
Gazebo version: 11.5
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Simbody_LIBRARIES (ADVANCED)
    linked by target "AircraftConfigPlugin" in directory /mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins
    linked by target "FlightControllerPlugin" in directory /mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins

-- Configuring incomplete, errors occurred!
See also "/mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/build/CMakeFiles/CMakeOutput.log".
See also "/mnt/d/PycharmProjects/attitude_control_gymfc_will/gymfc/envs/assets/gazebo/plugins/build/CMakeFiles/CMakeError.log".`
 the cmake log is given in this dropbox link:

https://www.dropbox.com/sh/rhxqmlu23e4gfyu/AAAcV5qSIaFUPcOCukHJkxhaa?dl=0