wil3/gymfc #103: error: Failed to compile Gazebo plugin

alexsandercaac, 2021-06-16:
### Description
The '_install_dependencies_.sh' executes without any error after updating "_logname_" in line 9, which was giving me an error, by "_whoami_" --- I believe their outputs should be equivalent.
When running the next command with pip3, I get the following error:
```
Processing /home/alex/workspace/gymfc
Collecting numpy (from gymfc==0.1.2)
  Using cached https://files.pythonhosted.org/packages/45/b2/6c7545bb7a38754d63048c7696804a0d947328125d81bf12beaa692c3ae3/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl
Collecting protobuf (from gymfc==0.1.2)
  Using cached https://files.pythonhosted.org/packages/53/4e/e2db88d0bb0bda6a879eea62fddbaf813719ce3770d458bc5580512d9c95/protobuf-3.17.3-cp36-cp36m-manylinux_2_5_x86_64.manylinux1_x86_64.whl
Collecting psutil>=5.3.0 (from gymfc==0.1.2)
Collecting six>=1.9 (from protobuf->gymfc==0.1.2)
  Using cached https://files.pythonhosted.org/packages/d9/5a/e7c31adbe875f2abbb91bd84cf2dc52d792b5a01506781dbcf25c91daf11/six-1.16.0-py2.py3-none-any.whl
Installing collected packages: numpy, six, protobuf, psutil, gymfc
  Running setup.py install for gymfc ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-jmpeqtxu-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-xohlp8m_-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    [gymfc] starting Gazebo plugin build.
    [gymfc] plugin directory located at /tmp/pip-jmpeqtxu-build/gymfc/envs/assets/gazebo/plugins
    -- The C compiler identification is GNU 7.5.0
    -- The CXX compiler identification is GNU 7.5.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    CMake Error at CMakeLists.txt:4 (find_package):
      By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
      asked CMake to find a package configuration file provided by "gazebo", but
      CMake did not find one.

      Could not find a package configuration file provided by "gazebo" with any
      of the following names:

        gazeboConfig.cmake
        gazebo-config.cmake

      Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
      "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
      provides a separate development package or SDK, be sure it has been
      installed.

    -- Configuring incomplete, errors occurred!
    See also "/tmp/pip-jmpeqtxu-build/gymfc/envs/assets/gazebo/plugins/build/CMakeFiles/CMakeOutput.log".
    make: *** No targets specified and no makefile found.  Stop.
    error: Failed to compile Gazebo plugin.

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-jmpeqtxu-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-xohlp8m_-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-jmpeqtxu-build/
```
### Steps to Reproduce
On a fresh install of Ubuntu 18.04, run apt-get update and upgrade. Next, substitute "_logname_" by "_whoami_" in the '_install_dependencies.sh_' file. Run the script without any issues. Run the _pip3_ command and see the error.

**Expected behavior**
Expected the installation of gymfc.

wil3, 2021-06-16:
Hi @alexsandercaac what was the error logname it was giving you? That is suspect, and has never been reported. Does the error get thrown if you don't do and upgrade? The dependencies are very sensitive to their versions which is why the script exists. The script does an update but not an upgrade so I suspect doing the upgrade caused a dependency issue. I find testing the install in a fresh virtual machine is useful.

alexsandercaac, 2021-06-16:
Thanks @wil3 for responding so quickly. The error with _logname was_: `logname: no login name`. I will repeat the installation _without_ upgrading apt-get, hopefully that will fix it! I will report here the outcome. I should mention, by the way, that I am using a WSL (Windows Subsystem for Linux) based Ubuntu 18.04, would you say that could be the cause?

CDM1619, 2022-10-13:
Hi,I have met the similar error when I run the pip3 install .  Could you help me? Great thanks for you!

  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in /home/chengdong/anaconda3/envs/flightmare/lib/python3.6/site-packages (from gymfc==0.1.2) (1.15.4)
Requirement already satisfied: protobuf in /home/chengdong/anaconda3/envs/flightmare/lib/python3.6/site-packages (from gymfc==0.1.2) (3.10.0)
Requirement already satisfied: psutil>=5.3.0 in /home/chengdong/anaconda3/envs/flightmare/lib/python3.6/site-packages (from gymfc==0.1.2) (5.6.3)
Requirement already satisfied: six>=1.9 in /home/chengdong/anaconda3/envs/flightmare/lib/python3.6/site-packages (from protobuf->gymfc==0.1.2) (1.12.0)
Requirement already satisfied: setuptools in /home/chengdong/anaconda3/envs/flightmare/lib/python3.6/site-packages (from protobuf->gymfc==0.1.2) (49.6.0.post20210108)
Building wheels for collected packages: gymfc
  Building wheel for gymfc (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/chengdong/anaconda3/envs/flightmare/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/chengdong/gymfc/setup.py'"'"'; __file__='"'"'/home/chengdong/gymfc/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-f7ur6zkr
       cwd: /home/chengdong/gymfc/
  Complete output (79 lines):
  running bdist_wheel
  running build
  [gymfc] starting Gazebo plugin build.
  [gymfc] plugin directory located at /home/chengdong/gymfc/gymfc/envs/assets/gazebo/plugins
  -- The C compiler identification is GNU 9.4.0
  -- The CXX compiler identification is GNU 9.4.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
  -- Checking for module 'bullet>=2.82'
  --   Found bullet, version 2.88
  CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/simbody/SimbodyConfig.cmake:131 (if):
    if given arguments:
  
      "TRUE"
  
    An argument named "TRUE" appears in a conditional statement.  Policy
    CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
    "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.
  Call Stack (most recent call first):
    /usr/local/lib/cmake/gazebo/gazebo-config.cmake:119 (find_package)
    CMakeLists.txt:4 (find_package)
  This warning is for project developers.  Use -Wno-dev to suppress it.
  
  -- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time
  -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1")
  -- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")
  -- Looking for OGRE...
  -- OGRE_PREFIX_WATCH changed.
  -- Checking for module 'OGRE'
  --   Found OGRE, version 1.9.0
  -- Found Ogre Ghadamon (1.9.0)
  -- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
  -- Looking for OGRE_Paging...
  -- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
  -- Looking for OGRE_Terrain...
  -- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
  -- Looking for OGRE_Property...
  -- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
  -- Looking for OGRE_RTShaderSystem...
  -- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
  -- Looking for OGRE_Volume...
  -- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
  -- Looking for OGRE_Overlay...
  -- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
[... 89 lines cut ...]
    -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "2.3.0")
    -- Config-file not installed for ZeroMQ -- checking for pkg-config
    -- Checking for module 'libzmq >= 4'
    --   Found libzmq , version 4.3.2
    -- Found ZeroMQ: TRUE (Required is at least version "4")
    -- Checking for module 'uuid'
    --   Found uuid, version 2.34.0
    -- Found UUID: TRUE
    -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1")
    -- Checking for module 'protobuf'
[... cut ...] p.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-1xuk_jvq/install-record.txt --single-version-externally-managed --compile --install-headers /home/chengdong/anaconda3/envs/flightmare/include/python3.6m/gymfc Check the logs for full command output.

njpzn, 2022-11-30:
I also have this probeles when I run the pip3 install . Could you solve me? 
  /home/pzn/gymfc/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp:68:10: fatal error: gazebo/physics/dart/DARTModel.hh: 没有那个文件或目录
       #include <gazebo/physics/dart/DARTModel.hh>
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      CMakeFiles/FlightControllerPlugin.dir/build.make:62: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o' failed
      make[2]: *** [CMakeFiles/FlightControllerPlugin.dir/FlightControllerPlugin.cpp.o] Error 1
      CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/FlightControllerPlugin.dir/all' failed
      make[1]: *** [CMakeFiles/FlightControllerPlugin.dir/all] Error 2
      Makefile:83: recipe for target 'all' failed
      make: *** [all] Error 2
      error: Failed to compile Gazebo plugin.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> gymfc

luoshenhua1, 2023-03-11:
> ### Description
> The '_install_dependencies_.sh' executes without any error after updating "_logname_" in line 9, which was giving me an error, by "_whoami_" --- I believe their outputs should be [...]
I encountered the same problem, how did you solve it at that time?