wil3/gymfc #105: Failed to launch quadcopter

NahomGebremeskel, 2021-07-25:
I have followed your installation tutorial, everything installed properly but gazebo failed to launch. 
What could be the problem?
I really need your help!
[image]

wil3, 2021-07-25:
That file path isn't used in the tutorial because it doesn't exist like the error is saying. I suggest watching it again, the digital twins are in the examples directory.

wil3, 2021-07-25:
From the look at the commands you are trying it appears you aren't familiar with the  Linux command line or python. It would be a good idea to do some basic tutorials such as paths and how to navigate and learn the argparse python library.

closed 2021-07-25 by wil3