wil3/gymfc #109: libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org

eislerMeinName, 2023-03-01:
### Prerequisites
* [ x] Put an X between the brackets on this line to verify you have done the following:
    * I have checked that your issue isn't already filed.
    * I am running Ubuntu 18.04
    * I installed the environment and GymFC according to these [instructions](https://github.com/wil3/gymfc/#quick-start). 
    
### Description
Hey,
I am currently trying to use GymFC for my bachelor thesis. I just installed it like shown in the video and used Gazebo11. Worked fine until trying out the test_step_sim script. When resetting the environment the script just hang up. After some time i got the already mentioned report: libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
I debugged it into line 10 of the code, where the environment is reset, but did not dive further into it. 
At the moment i am using Arch Linux, but i also got the same mistake when using it on an Ubuntu 18.04 virtual machine.
I hope you can help me with this problem.

### Steps to Reproduce
start env, create sim...
1. Execute command X
ob = env.reset()
Output from command X
libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org

wil3, 2023-03-07:
Hi eislerMeinName,

Its been a long time since I looked at this project but my guess is Gazebo hasn't done a good job with backwards compatibility. Googling the issue it seems like a couple other people are having similar issues but no solution. What strange is there is an A record for that domain but I can't ping it. 

Have you debugged where in the gazebo plugin is causing that call?  If its happening from reset that occurs [here](https://github.com/wil3/gymfc/blob/master/gymfc/envs/assets/gazebo/plugins/FlightControllerPlugin.cpp#L620). Uncommenting some of those debug statements may help to isolate which call is for whatever reason making a curl request. I'm wondering if its trying to grab a schema or something. 

Something else that may help is looking at [this](https://gazebosim.org/api/fuel_tools/5.0/cmdline.html) resource. Maybe its something from within Gazebo.

eislerMeinName, 2023-03-29:
Hey, sorry for the late answer. I realized that the problem occurs when there is a timeout when communicating with the flight control plugin. Reinstalled everything and did a new vm, now it works.

wil3, 2023-03-29:
Awesome! Glad to hear.